Ultimaker/Cura #9653: Application crashes when opening STL files exported from Blender 2.90.1

thedaviator, 2021-04-22:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Open Cura 4.9 
Open any STL file exported from Blender 2.91.0
Application immediately crashes:

**_Error traceback_**

"Traceback (most recent call last):
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'

Logs:
Thread 0x0000286c (most recent call first):
  File "threading.py", line 306 in wait
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1496 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002e8c (most recent call first):
  File "threading.py", line 306 in wait
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1315 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000a00 (most recent call first):
  File "C:\Users\xxxxxx\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1221 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001c6c (most recent call first):
  File "ntpath.py", line 233 in islink
  File "serial\tools\list_ports_common.py", line 48 in __init__
  File "serial\tools\list_ports_windows.py", line 209 in iterate_comports
  File "serial\tools\list_ports_windows.py", line 299 in comports
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 118 in getSerialPortList
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 85 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000c18 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000a1c (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
[... 150 lines cut ...]
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

### Actual results

App crashes no matter which file I attempt open. All files tested opened  and sliced fine in Cura 4.7 

### Expected results

I expected you to fix the sizing issue for Blender-created STLs that was introduced in v 4.8. Instead, you made it worse. 

### Additional information

Who does your QA? Was this release even tested with Blender created STLs?  Now I have to reinstall v 4.7... again.

fvrmr, 2021-04-22:
Hi @thedaviator thank you for your bug report.
Could you provide a file, so I can try to reproduce it.
Thank you in advance.

nallath, 2021-04-22:
You need to update the the custom prefix plugin. An update has been available for +- 1 month already. 

If you want to help out with the QA, feel free to download our Beta releases that we provide +-2 weeks before every release! I'm sure that with your help we will be able to catch issues like this much faster. As you might imagine, we don't test all plugin combinations out there.

closed 2021-04-22 by nallath

thedaviator, 2021-04-22:
Sure,

Thanks,
David

fieldOfView, 2021-04-23:
Unfortunately github strips attachments from email replies. Could you please post a link, or attach the file via https://github.com/Ultimaker/Cura/issues/9653 ?

thedaviator, 2021-04-23:
here you go
[ladder guide.zip](https://github.com/Ultimaker/Cura/files/6365126/ladder.guide.zip)