Ultimaker/Cura #9656: Final layer printing temp is ignored

dmitche3, 2021-04-22:
### Application Version

4.8-4.9

### Platform

Windows 10

### Printer

Creality 3 V2

### Reproduction steps

This setting doesn't do anything.  Final layer temperature never changes.  

### Actual results

As described above.

### Expected results

As described above.

fvrmr, 2021-04-22:
Hi @dmitche3 thank you for your bug report.
Also for this issue could you provide me a project file (file - save project).
Thank you in advance.

dmitche3, 2021-04-23:
LOL. It didn’t do anything on any project, ever. It don’t make a difference.

fvrmr, 2021-04-23:
Do you have any reproduction steps to reproduce this?

dmitche3, 2021-04-23:
I examined the G-Codes being created and they appear to be correct according to the definitions for the G-Codes, yet the Ender 3 V2 is not treating them properly.
For example, the following code will wait to warm the bed to 60C AND THEN warm the hotend.  It doesn't warm both at the same time.  According to the website https://www.makeit-3d.com/wp-content/uploads/RepRapGcodeCheatSheet.pdf when setting the M140 S60 it shouldn't wait.  Perhaps the M105 query is causing a wait? 

;Generated with Cura_SteamEngine 4.9.0
M140 S60
M105
M190 S60
M104 S225
M105
M109 S230

With regards to the final print temperature I see it being set there appears to be no G-code generated.

[New Plates Final GOLDEN.zip](https://github.com/Ultimaker/Cura/files/6365911/New.Plates.Final.GOLDEN.zip)

GregValiant, 2021-04-24:
The M190 S60 is before the M104 and so it waits.  Then it waits for the hot end to hit 230.  When that is the case both temps have to be met before printing will start.  An additional factor would be that not all printers have power supplies robust enough to provide full power to both devices for heatup.  They can float two at the set temperatures, but not heat both together.

If you have the gcode tell the printer to heat the bed and immediately heat the hot end (and wait for the hot end) then the print will start when the hot end hits 230 and it won't care what the bed temp is.  It would kind of defeat the purpose of having an Initial Bed Temperature set.  Without the wait, you've told the printer you don't care.
 
If you have the "Printer Settings" plugin loaded you can de-select "Wait for build plate heatup".  That will eliminate the M190 line.

dmitche3, 2021-04-24:
I’ll look into this when I can. I was hoping that the software would have sent out the initial settings so that both the bed and hidden hotend would begin to heat, and then send the code to wait for the bed temp and then gotten to reach the proper temperature.

GregValiant, 2021-04-24:
If you use Cura keywords in your Start-Up Gcode then Cura will replace the keywords with the actual values.  A side effect is that it will not put those temperature lines in front of the start gcode.  The Keywords are [HERE](http://files.fieldofview.com/cura/Replacement_Patterns.html).  The following are some examples.
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp

When doing it that way it results in gcode like this:
;Generated with Cura_SteamEngine 4.9.0
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M104 S210 ; set extruder temp
M140 S50 ; set bed temp
M109 S210 ; wait for extruder temp
M190 S50 ; wait for bed temp

That would mean a lot of power at startup and is one reason why the M109 and M190 commands are used.  The other is that the wait commands insure printing won't start until the temps have been met.

Ghostkeeper, 2021-05-10:
> LOL. It didn’t do anything on any project, ever. It don’t make a difference.

If you've ever used Cura you've used a "Cura project" too, even if you never saved the project to disk. A project is the state o the models, profiles and settings you've been using. We're asking simply to save it to disk and share that file with us so that we can reproduce the problem. That's under File -> Save...

There is no "final layer printing temperature" setting. There is a Final Print Temperature setting though, so I'll assume you meant that. You didn't provide any reproduction steps either, but I can prove that the setting does have an effect. These are two project files with different Final Print Temperature settings: [final_print_temp.zip](https://github.com/Ultimaker/Cura/files/6452186/final_print_temp.zip)

Slicing these and finding the difference reveals that 82 lines of g-code have changed, all of the form:
[image]

The Final Print Temperature has no effect on the start g-code (unless modified) and no effect on single-extruder prints like the one you provided the g-code of. So this seems to me like the original issue in this thread is just a simple misunderstanding of the Final Print Temperature setting, and not an actual issue. I've written a more detailed description of the setting here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/material/material_final_print_temperature.md

closed 2021-05-10 by Ghostkeeper