Ultimaker/Cura #9657: It's Ironing the top -INFILL- layer.

batista16, 2021-04-22:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3 pro

### Reproduction steps

I've activated ironing, and it's ironing not only the top layer, but also the top infill layer. 

### Actual results

it's ironing not only the top layer, but also the top infill layer. 

### Expected results

It should iron only the top layers. 

### Additional information

Layer 6
[image]

[image]
Layer 7
[image]

[image]
Layer 8
[image]
[image]

Project file
[Molde Exterior.zip](https://github.com/Ultimaker/Cura/files/6359397/Molde.Exterior.zip)

fvrmr, 2021-04-23:
Hi @batista16 thank you for your bug report. 
It looks like it has to do with the infill mesh. If you change it to a cutting mesh it won't happen. Let me know if that works for you.

batista16, 2021-04-23:
Hi fvrmr, yep, that fixed it. 
But it's still a bug right? or is it a feature that I don't understand?

Ghostkeeper, 2021-05-10:
It's indeed a feature you didn't properly understand. And in fact I think fvrmr was wrong in the sense that a cutting mesh still causes ironing around that region. It just won't be on top of infill, but on top of the skin that appears in the cutting plane when you use a cutting mesh.

You're using an infill mesh there. An infill mesh is effectively just a normal mesh that shapes itself to the infill regions of other meshes, which allows you to edit that region of infill in a myriad number of ways.

Normally the infill mesh is automatically set to have 0 walls and 0 top/bottom layers. This makes the infill mesh "invisible" because without walls and skin, all that's left is infill. So the infill mesh replaces infill with infill. That is done so that it's easy for people to change the infill density in a certain region, the most common application for infill meshes. You've changed the Wall Line Count to 3 though, so the infill region now gets 3 extra walls (effectively giving your model 6 walls in that lower region).

However the problem is that you've enabled coasting here. Ironing is applied to the top side of every mesh. That includes the infill mesh. This could be nice if you're doing wicked things but usually not so much. Since your infill mesh still has no top layers, the top of the mesh is an exposed infill pattern. So you get ironing on top of the infill pattern, somewhere in the middle of your infill.

So what you could do is add the "Enable Ironing" setting to the per-object settings of the infill mesh and then disable it:
[image]

As far as I'm concerned, this is really more of a UX issue. How could we make this behaviour more clear or easier to use?

batista16, 2021-05-10:
I totally understood the way it works now, thanks for that. 

About making it more clear, it's honestly an hard one.. 
I would do for ironing the same as you do for Wall/Top/Bottom thickness, automatically turn it off and add the overlap setting in there. that way would be clear that the ironing feature was turned off automatically, and if my any weird reason someone want's it on, they would have to change it yet.. I think it's much more common to want it OFF in overlaps then ON, so I think it would benefit most use cases. 

Although I think it's a good solution, it opens the precedence for other settings.. probably there would have to be a study on which settings to automatically change for overlaps..

batista16, 2021-05-10:
BTW, the issue can be closed if you want, or you can transform it into an UX improvement ticket..

fvrmr, 2021-05-11:
We have discussed this issue with the team, and decided to defer the UX issue. It is not a priority to improves this right now. 
Hope you understand.