Ultimaker/Cura #9659: inconsistency with decimal from integer part separation

Liger0, 2021-04-22:
### Application Version

4.9

### Platform

Windows 10

### Printer

Any

### Reproduction steps

Look at the profile and the materials

### Actual results

In the profile, the decimals use the dot, while in the materials they use the comma.

### Expected results

Unified behaviour. Either using the dot like the computer does or the comma like humans do.

jellespijker, 2021-04-22:
This is dependent on you system locale. The fields in Cura should use the one specified by your system ( most likely as custom in your country), while the profile files can be considered a part of source code. Which is developed using the en_US locale. These specify a dot as decimal separator. Which is often used by developers.

Changing how we store values in the profile files such that they follow the system locale of the user is imo more work then it's worth.

Could you maybe explain me what you think will be the benefits for this?

Liger0, 2021-04-22:
I'm not talking about profile files but the profile GUI vs material GUI.

jellespijker, 2021-04-22:
Okay, thanks for the clarification. That should be uniform imo opinion as well. I will discuss it with my colleagues tomorrow morning.

Ghostkeeper, 2021-05-11:
This was added to an existing issue on our backlog, CURA-5955, the ticket to get rid of the system-styled elements in the preferences screens. Replacing the system-generated text boxes with our own versions will fix this issue.

The system-styled text boxes do proper locale and such which is nice to non-technical personnel, but it's inconsistent with the rest of Cura and also inconsistent with how many engineers work.

closed 2024-12-08 by github-actions[bot]