Ultimaker/Cura #9660: Cura 4.9 windows 10 crashes shortly after opening

AllenF88, 2021-04-22:
### Application Version

4.9

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. I installed cura 4.9 without uninstalling 4.8. Cura 4.8 and 4.9 both crashed as it opens. I uninstalled both and just installed 4.9, it opens but as soon as I click on Get started, it closes immediately, no errors. Error log attached 
[stderr.log](https://github.com/Ultimaker/Cura/files/6360691/stderr.log)

### Actual results

Cura 4.9 closes without an error dialog.

### Expected results

It should work

fvrmr, 2021-04-23:
Hi @AllenF88 thank you for your report.
I will discuss this with the team. Keep you posted!

coolgua888, 2021-04-24:
I also have the same problem. After opening it, click anywhere and the program will not respond and crash

haithamkob, 2021-04-24:
am also having this problem, when starting 4.9 crashes and gives this msg ( see pic), 4.8 working well.
am using Windows 10 pro.
[image]

lucafedechen, 2021-04-25:
hi
same error here! attached log
launch the program and crashes
thanks
[stderr.log.txt](https://github.com/Ultimaker/Cura/files/6371154/stderr.log.txt)

AllenF88, 2021-04-25:
> Hi @AllenF88 thank you for your report.
> I will discuss this with the team. Keep you posted!

Thanks @fvrmr. I saw window events saying my OpenGL was corrupted or couldnt be found. Not sure if it was caused by installing the 4.9 update.

LAP87, 2021-04-25:
Why is cura looking for X: drive at all?
I'm using X: for a mounted cloudstorage, and cura 4.9 crashes at boot with this message.
[image]

alexdelli, 2021-04-29:
the same here

Ghostkeeper, 2021-05-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/9600 . This is fixed in version 4.9.1, which is to be released shortly.

closed 2021-05-11 by Ghostkeeper