Ultimaker/Cura #9663: Cura 4.9 not installing

LouwrensMB, 2021-04-23:
### Application Version

4.9

### Platform

windows 10

### Printer

Creality CR6 SE

### Reproduction steps

Downloaded the file from github and installed it while installing I get a ERROR opening file for writing 

### Actual results

Give me a error opening file

### Expected results

abort

### Additional information

any help

LouwrensMB, 2021-04-23:
I renamed the file and it worked

fvrmr, 2021-04-23:
Hi @LouwrensMB great to hear that it is working now. 
Just curious what file did you rename?

LouwrensMB, 2021-04-24:
changed the file name to Ultimaker Cura 4.9.0 Final

closed 2021-04-29 by fvrmr

evtrados, 2021-05-17:
Tried this on multiple Windows 10 machines, but I was not able to reproduce it, unfortunately.