Ultimaker/Cura #9664: Live print feature

abaucom85, 2021-04-23:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

Please add a feature that would allow users to view the layer/position of the model during the print ('live feed' type printing. In other words, when the user is printing (from their computer directly to their printer), show a model of the print that is taking place so the user would know at what point during the print it is at, aside from only the time remaining, etc. Like how the preview tab allows the user to be able to step it back layer by layer, and play out how the nozzle would be printing, integrate that into the "live/active print" section to allow the user to see how the print job is actively doing, what step/layer/xyz position it is at, etc.
I believe this would help play an integral role in how the users can best utilize what is actively playing out while their design is being printed. I know that for me, it would be a great asset to be able to use during the live print.

### Describe alternatives you've considered

I have not considered any alternatives yet

### Affected users and/or printers

I believe this would affect all users/ possibly printers. I have the Anycubic Mega Pro printer, and believe this feature could be very helpful.

fvrmr, 2021-04-23:
Hi @abaucom85 thank you for your feature request.
I will discuss this with the team. Keep you posted.

fvrmr, 2021-04-26:
I have discussed it with the team. And decided to defer this feature, because it is not one of our priority's right now. It is also a duplicate of https://github.com/Ultimaker/Cura/issues/6020
Hope you understand!

closed 2021-04-26 by fvrmr

abaucom85, 2021-04-26:
My priority will now then be to use another program and defer from Ultimaker. The way your email was stated was a bit rude, so I will take my business elsewhere. Hope you understand!

Sent via the Samsung Galaxy S20+ 5G, an AT&T 5G smartphone