Ultimaker/Cura #9667: Overlaid displaying of numbers

Ramzor-1, 2021-04-23:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

FBG5

### Reproduction steps

Just install the latest version 4.9.0 
Bad visual displaying of numbers. Looks like they are overlaid on each other.
See attached file

[image]

### Actual results

Just install the latest version 4.9.0 
Bad visual displaying of numbers. Looks like they are overlaid on each other.
See attached file

### Expected results

Bad visual displaying of numbers. Looks like they are overlaid on each other.

jellespijker, 2021-04-23:
This might be a duplicate of #9643

happysleeper, 2021-04-24:
I have the same annoying problem

fvrmr, 2021-04-26:
Hi @Ramzor-1 it looks like a duplicate indeed of #9643 but I would like to keep this open since you've posted screenshots.
Could you also provide me a log file? Maybe we can find something there.

SnowflakeSolutions, 2021-04-26:
I have the same Issue, attached the log file
[cura.log](https://github.com/Ultimaker/Cura/files/6378860/cura.log)

fvrmr, 2021-04-30:
I can reproduce this issue when I set the language of my computer to a RTL language like Arabic. I will discuss this issue with the team. Keep you posted!

fvrmr, 2021-05-03:
I have discussed it with the team. And we've created a ticket on our backlog.
Developers see CURA-8206.

commit 836e7c0c (2021-06-07): Fix overlap of unit and setting value with RTL-languages

The unit was hard-coded to be anchored to the right. The text was getting the full width and aligned implicitly by the language. We'll keep the layout of the text properly internationalised, but make sure that the unit is opposite of wherever the text is.

Fixes CURA-8206.
Fixes #9667.
closed 2021-06-07 by Ghostkeeper