Ultimaker/Cura #9668: Cura 4.9 crashes when using windows search

MarvinWalther, 2021-04-23:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Had 4.8.0 installed
2. Installed 4.9.0 (uninstalled old version but kept user setting)
3. Open Cura
4. Click on "open file" button
5. Type something (that exists / can be found) in the search bar
6. Hit enter

### Actual results

Cura crashes after showing search results for about 2 seconds.
But only if there are results to show. If I search for something, I know doesn't exists, it shows "no results" and everything is fine.

### Expected results

Show me search results and be able to open files.

### Additional information

In Windows file explorer, searching for files does work without problems. Tested with the same search in the same folder.
Also just open a file in Cura without searching for it does work.

### Log:
```
2021-04-23 13:56:38,399 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-04-23 13:56:38,409 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-04-23 13:56:38,416 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-04-23 13:56:38,421 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-04-23 13:56:38,426 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-04-23 13:56:38,432 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-04-23 13:56:38,445 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-04-23 13:56:38,452 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-04-23 13:56:38,461 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-04-23 13:56:38,466 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-04-23 13:56:38,473 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-04-23 13:56:38,480 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-04-23 13:56:38,486 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-04-23 13:56:38,491 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-04-23 13:56:38,498 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-04-23 13:56:38,505 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-04-23 13:56:38,513 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-04-23 13:56:38,518 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-04-23 13:56:38,523 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-04-23 13:56:38,531 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
2021-04-23 13:56:38,536 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-04-23 13:56:38,552 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimulationView
2021-04-23 13:56:38,559 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SliceInfoPlugin
2021-04-23 13:56:38,567 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-04-23 13:56:38,575 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-04-23 13:56:38,601 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-23 13:56:38,603 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-23 13:56:38,604 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-23 13:56:38,606 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-23 13:56:38,607 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-23 13:56:38,609 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-04-23 13:56:38,610 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-04-23 13:56:38,612 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
[... 101 lines cut ...]
2021-04-23 13:56:42,948 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 465.89
2021-04-23 13:56:42,949 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2021-04-23 13:56:42,951 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: NVIDIA GeForce GTX 1080/PCIe/SSE2
2021-04-23 13:56:42,953 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2021-04-23 13:56:42,954 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\default.shader]...
2021-04-23 13:56:42,957 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\selection.shader]...
2021-04-23 13:56:42,960 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\select_face.shader]...
2021-04-23 13:56:42,962 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\default.shader]...
2021-04-23 13:56:42,965 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\composite.shader]...
2021-04-23 13:56:42,968 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\overhang.shader]...
2021-04-23 13:56:42,971 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\striped.shader]...
2021-04-23 13:56:42,974 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\transparent_object.shader]...
2021-04-23 13:56:42,977 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\striped.shader]...
2021-04-23 13:56:42,980 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\xray.shader]...
2021-04-23 13:56:42,982 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\xray_composite.shader]...
2021-04-23 13:56:42,985 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\default.shader]...
2021-04-23 13:56:42,988 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\grid.shader]...
2021-04-23 13:56:42,991 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\platform.shader]...
2021-04-23 13:56:43,008 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.9.0\resources\shaders\xray.shader]...
2021-04-23 13:56:43,083 - DEBUG - [JobQueWorker [2]] USBPrinting.AutoDetectBaudJob.run [49]: Checking COM6 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
2021-04-23 13:56:45,133 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 4.0s
2021-04-23 13:56:47,608 - DEBUG - [JobQueWorker [2]] USBPrinting.AutoDetectBaudJob.run [49]: Checking COM6 if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
2021-04-23 13:56:52,127 - DEBUG - [JobQueWorker [2]] USBPrinting.AutoDetectBaudJob.run [49]: Checking COM6 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2021-04-23 13:56:53,739 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
```

fvrmr, 2021-04-26:
Hi @MarvinWalther thank you for your bug report. 
I can't reproduce this issue. But it seems like this might be related to some other issues.
I will discuss this with the team keep you posted.

evtrados, 2021-04-29:
I can only reproduce this issue if:

- I have Chitubox installed. Doesn't matter if it's currently running or not or whether you enable the thumbnail preview during the installation
- **AND** I use the Preview Pane on Windows Explorer. If I disable the latter then Cura no longer crashes. It will crash again though if you try and select a different View option (e.g., large icons, etc). 

The only reason I can think of that this happens on Cura 4.9 and not on 4.8 is because we now use Qt 5.15.2 and Chitubox 1.8.1 uses Qt 5.15.1. I've had similar issues on macOS where two different applications that were using the same Qt version were crashing because the OS would find the same library in different locations in the system.

nallath, 2021-04-29:
So it's actually chitubox "poisoning" the operating system by inserting their version of QT into the preview dialog.

CBDesignS, 2021-05-03:
This has bugged me for a few days since updating, I had to move files into the last opened folder because of this untill it pi$$ed me off enough to dig into it and evtrados is correct. Chitucrap is screwing things over. I never used chitubox64 1.6.4.3.beta for a long time but removing the dll sorted the crash instantly.  regsvr32 /u "C:\Program Files\ChiTuBox64 1.6.4.3 Beta\CHITUBOX_Thumbnail.dll". Thnks for the heads up

inlsadler, 2021-05-15:
This is helpful also have Chitubox installed which I like for my resin printer.  If I disable the DLL will Chitubox still work?

CBDesignS, 2021-05-15:
@inlsadler it should work ok. I have never used my resin printer for over a year and I am dreading looking inside as I left resin in the tray. I could have a fubar machine but on the original question. Shitubox still opend and slices so I see no reason why it would not work.. I think this dll problem has been round for years now so disable it and get back to printing again.

inlsadler, 2021-05-15:
All sorted and both working now.  Thanks for the help.

chepo92, 2021-05-16:
Same here, with 4.9 and 4.9.1, but I don't have shitubox installed

Ghostkeeper, 2021-05-17:
Perhaps you could share a log file then, @chepo92?

chepo92, 2021-05-17:
Attached log file, I just opened Cura and drag n' drop a file from desktop

[cura.log](https://github.com/Ultimaker/Cura/files/6495352/cura.log)

fieldOfView, 2021-05-17:
@chepo92, please update the Printjob Naming plugin in the Marketplace.

chepo92, 2021-05-17:
> 
> 
> @chepo92, please update the Printjob Naming plugin in the Marketplace.

That worked for me! thks

fieldOfView, 2021-05-17:
For future reference: try to keep your plugins up to date for issues such as this. The update to this plugin was released more than a month before Cura 4.9, to prevent this exact issue.

nallath, 2021-05-27:
I'm going to close this issue as it's chitubox doing this (or a user needing to update plugins).

closed 2021-05-27 by nallath

robhp81, 2021-07-02:
The instructions you provided do not work!!!

Ghostkeeper, 2021-07-07:
In one of the duplicates, @3Duke3 asked if it's possible for Cura to create its own thumbnailer:

> Excuse my lack of deeper software understanding, but seeing as Cura is such a major STL application, would there be an option to develop your own STL thumbnail tool that took priority over shitubox?

We could probably develop a thumbnailer (if given time), but I don't think it's possible to ensure that it takes priority over Chitubox' extension. You can't just tell Windows that your program is more important than another one. I'd expect that it just depends on whichever extension you installed last. And if it is possible, having both extensions installed would still cause it to crash anyway, since Chitubox' Qt 5.15.1 DLL is still loaded by the operating system when the file dialogue is opened.

You can try it for yourself by installing an alternative thumbnailer. I've been using this one myself: https://github.com/unlimitedbacon/stl-thumb

Ghostkeeper, 2021-11-19:
We are working on the upgrade to Qt 6.2. It's currently in a bit of a preparation state still, not so far yet. The aim is to complete the upgrade for Cura 5.0 but when 5.0 will be released is uncertain.

This upgrade is expected to hide the issue. It's currently crashing because of a collision between two 5.15 versions of Qt, which link to the same libraries. Once Cura updates to 6.2, they should no longer collide. At least not until Chitubox also updates to 6.2.