Ultimaker/Cura #9669: cx python error

brendanssound, 2021-04-23:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Had version 4.8.0 
Ran installer for version 4.9.0 which asked if i wanted to remove old version.. I did.. So it did.. 
Attempted to run 4.9.0 

### Actual results

Get an error 

Error in sys.excepthook
Exception (WinError 53) The network path was not found.
x:\\4.9.exe\build\inst\bin\cura_app.py
Original Exception (WinError 53) The Network path was not found
x:\\4.9.exe\build\inst\bin\cura_app.py

No idea why its looking on a network drive for a file.. ?? 

Installed from my local drive.. never run anything off my network drives.. Don't use onedrive/any other cloud storage
Have removed the cura directories in Local & Roaming
Uninstalled/Reinstalled
Even restarted after uninstall and tried reinstall

### Expected results

Program runs, I get to slice prints and live happily ever after

brendanssound, 2021-04-23:
[image]

jellespijker, 2021-04-23:
Duplicate of #9647 will close this issue in favor of the discussion there.

closed 2021-04-23 by jellespijker