Ultimaker/Cura #9670: Automatic tool selection during starting G-Code crashes printer

Scissors95, 2021-04-23:
### Application Version

4.9

### Platform

Windows 10

### Printer

E3D Toolchanger

### Reproduction steps

Set up machine in cura, added file, sliced, studied GCode

### Actual results

first line of GCode is T0 then M98 Phomeall, called before custom starting Gcode is called. If the program is ran as-is the best case is that the printer would error out due to not being homed, the worst case is that the machine would crash because it cannot home with a tool selected.

### Expected results

First line of GCode is the custom starting GCode in printer settings.

Scissors95, 2021-04-23:
project file and GCode are attached

[Bicolor Cube Cura.zip](https://github.com/Ultimaker/Cura/files/6368031/Bicolor.Cube.Cura.zip)

fvrmr, 2021-04-26:
Hi @Scissors95 thank you for your report.
I will close this issue since it is related to #7395 and I will discuss that issue again with the team. I will post an update there.
If you have more ideas or questions you can also post them there.

closed 2021-04-26 by fvrmr