Ultimaker/Cura #9672: Cura 4.9.0 won't print 2 colors

engpete, 2021-04-23:
### Application Version

4.9.0

### Platform

Mac OS Big Sir

### Printer

Ultimaker S3

### Reproduction steps

After installing Cura 4.9.0 I was no longer able to print with 2 colors on my S3. I imported 3 parts, selected appropriate color PLA, then merged them for printing.

### Actual results

Print core 2 went through the motions of printing but no filament came out. Spencer Priddy from Ultimaker support tried printing my file with the same results on v4.9.0. When I reinstalled v4.8.0 I was able to print again.

### Expected results

Ability to print with 2 colors in a merged part file.

jellespijker, 2021-04-23:
Please attach the project file and log file. We need more information.

engpete, 2021-04-23:
Here is the project file. Note: attachments are from v4.8.0. I don’t know how to get a log file.

Best,
Pete

jellespijker, 2021-04-23:
Github doesn't accept files when sending via email. Could you add them via the browser?
For log files see this link https://github.com/Ultimaker/Cura#logging-issues

engpete, 2021-04-23:
Github doesn't support the Project file type, so I zip filed it.
[UMS3_1c829b68-6f70-4343-9490-2ea702219057.3mf.zip](https://github.com/Ultimaker/Cura/files/6367935/UMS3_1c829b68-6f70-4343-9490-2ea702219057.3mf.zip)

I am on a Mac and your instructions for log file is for windows. Where do I find the log file on Mac OS?

jellespijker, 2021-04-23:
It's written there:
```
$USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
```
But I agree it could be written down a bit more clearly.

You can also check the help menu, there should be an option named configuration location or something. This should open up the file browser where the log file is located.

engpete, 2021-04-23:
Understood, but there is nothing there. See attached. I will get the log files from the printer via USB stick and upload it to you.

engpete, 2021-04-23:
Here are the attached.
[Printer logs.zip](https://github.com/Ultimaker/Cura/files/6368198/Printer.logs.zip)
[Mac OS.zip](https://github.com/Ultimaker/Cura/files/6368200/Mac.OS.zip)

jellespijker, 2021-04-23:
Slicing your model is generating GCode. In that GCode I can find the appropriate `T0` and `T1` commands which indicate tool changes. I can also see `E` values on `G1` commands. This indicates that the GCode generated by Cura is correct.
I believe it might be something hardware related.

Here is a snippet of the generated code
```
T1
G92 E0
M109 S200
M104 T0 S100
G1 F1500 E-6.5
G1 F600 Z2.2
G0 F1285.7 X160.324 Y110.236 Z2.2
M204 S500
M205 X5 Y5
;TYPE:SKIRT
G1 F600 Z0.2
G1 F1500 E0
G1 F600 X159.446 Y110.475 E0.01198
G1 X158.056 Y110.747 E0.03063
G1 X156.921 Y110.874 E0.04567
G1 X156.65 Y110.889 E0.04924
G1 X154.716 Y110.928 E0.07471
G1 X114.143 Y110.926 E0.60895
```
Are you sure it isn't hardware related? Maybe a clogged nozzle or grinded filament?

engpete, 2021-04-23:
Definitely not hardware related. The problem started after upgrading to v4.9.0. The problem went away after going back to v4.8.0. Furthermore, I sent my project file to Spencer Priddy (servicedesk@ultimaker.com) from Ultimaker Customer Service and he had the same issue when printing from his printer. See attached reply from Spencer.
[image]

jellespijker, 2021-04-24:
Thank you for letting me know. I will test your project file on my personal UM3 this weekend and I will ask @rburema if he can test in on his S3. But that will probably somewhere mid-week. Since we got a national holiday on Tuesday. I will keep you informed.

Ultiflow, 2021-04-26:
I don't know if it is that issue, but when a do a dual extrusion with Cura 4.9 (first time I try), the first layers is not printed in the "other material"

I have a UM5 pro bundle

If I make a skirt with material 1, then the 2 is not printed on first layers, if I make a skirt with material 2, it's material 1 that is not printed, the filament is seen in the bowden and the extruded is working in emptiness.

I have tried with or without dual extrusion prime tower. 

(ten times)

jellespijker, 2021-04-26:
Devs see MISP-1388

billwadetec, 2021-04-26:
I have this same issue with PLA and PVA.
The second material does not extrude on the first layer.
4.8 works fine.
I have replicated this on 2 different s3's and an S5.
It looks like the feeder settings for the first layer are incorrect.

Dim3nsioneer, 2021-04-26:
I can confirm the issue for both S3 (with firmware 6.3) and S5 Pro Bundle (with firmware 6.2). Also, second material extrusion works on the S3 with the same project sliced in Cura 4.8.
If one pushes the filament manually into the second extruder right when that one starts, one can push it for about 20mm. Also, if you don't do anything manually, it will start to extrude the material at some (way too late) point.

nallath, 2021-05-06:
It was indeed an issue with Cura. We've fixed it for 4.9.1 that will be released soonish. 

For more (technical information about it and how we fixed it) you can have a look at https://github.com/Ultimaker/CuraEngine/pull/1445 and https://github.com/Ultimaker/CuraEngine/pull/1447

closed 2021-05-06 by nallath