Ultimaker/Cura #9673: Cura 4.9 Locks up Mac

Bflat703, 2021-04-23:
### Application Version

4.9.0

### Platform

Mac OS 10.13.6

### Printer

CR-10

### Reproduction steps

Slice Project
Switch to "Preview" mode

First attempt locked up the entire computer.
Second attempt locked up Cura.
[Reflecting Unit Problem.3mf.zip](https://github.com/Ultimaker/Cura/files/6368269/Reflecting.Unit.Problem.3mf.zip)

### Actual results

System locked up.
Force restart.

### Expected results

Sort of expected it to work.

### Additional information

Mac Pro (Mid 2010) 3.2 GHz Quad-Core Xeon 32GB
ATI Radeon HD 5770 1024 MB

[cura.log](https://github.com/Ultimaker/Cura/files/6368283/cura.log)

talondnb, 2021-04-24:
Same here. iMac 2011, complete hangup but if I click another tab quick enough (e.g. Monitor or Prepare) I can avoid the full lockup.

Vasiy, 2021-04-24:
same issue

jellespijker, 2021-04-24:
Duplicate of #9631 I will therefore close this issue in favor of that one.

closed 2021-04-24 by jellespijker