Ultimaker/Cura #9674: [4.9] Sporadic crashes when opening folders to load a file

Liger0, 2021-04-23:
### Application Version

4.9.0

### Platform

Windows 10 x64

### Printer

Cr10S

### Reproduction steps

1. File
2. Open file
3. Navigate into directories

### Actual results

At a certain point, when opening the last folder with the stl I needed, Cura crashed.
I reopened it, and it crashed at the same directory
At the third try it didn't anymore....

### Expected results

No crash?

Liger0, 2021-04-23:
PS: the new github layout doesn't work correctly: if you attach an attachment it removes it and you have to edit it through the old layout to insert it.

jellespijker, 2021-04-24:
Thanks for letting us know @Liger0, that might explain why I need to ask for log files with practically every issue.
@fvrmr can you take a look into this

fvrmr, 2021-04-26:
@Liger0 thank you. I have updated the template!

I think that this is a duplicate of #9668 so I will close this issue and keep everything at one place.

closed 2021-04-26 by fvrmr

Liger0, 2021-04-26:
> 
> 
> @Liger0 thank you. I have updated the template!
> 
> I think that this is a duplicate of #9668 so I will close this issue and keep everything at one place.

I'm not sure if my issue is related to QtCore5.dll , is it?

evtrados, 2021-04-29:
> I'm not sure if my issue is related to QtCore5.dll, is it?
> In this case #9648 may also be related to #9688

@Liger0 , I think it is. You could verify that by causing Cura to crash and then opening the Windows Event Viewer --> expand 'Windows Logs' on the left --> select 'Application'. The crash will probably be the latest Error event and if you select that it'll say which module crashed.
For me it always mentions: `Faulting module name: Qt5Core.dll, version: 5.15.2.0, time stamp: blahblahblah`

In any case, I figured another alternative one could use to bypass this. For me the crash happens only when having Chitubox installed and I use the file preview mode on Windows Explorer. If I disable the latter then Cura no longer crashes.