Ultimaker/Cura #9675: Stops Printing and Shows Temperature Warning

DrFailbucket, 2021-04-23:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

I Sliced my Model with my Settings that i know it is working. I alredy Printed this Model 3 times without any issue, sliced with 4.8.0.  

I was changing only one Setting: from Skirt to Brim.

After doing the same change in the setting with Cura 4.8.0 it worked fine again

### Actual results

The printer starts and put down the first Layer. It moves to Z location for Layer 2 then it moves 5-10mm in X, Stops Printing, Starts Beeping and Shows the Temperature Warning on the LCD

### Expected results

I expected to be printed like the other 3 ones.

### Additional information

Here you can find the Project Files, and also the Sliced GCode Files

[Sliced_Files.zip](https://github.com/Ultimaker/Cura/files/6368451/Sliced_Files.zip)

DrFailbucket, 2021-04-23:
I forgot the Log files, so here they are. 

[cura_4.8.0.log](https://github.com/Ultimaker/Cura/files/6368490/cura_4.8.0.log)
[cura_4.9.0.log](https://github.com/Ultimaker/Cura/files/6368491/cura_4.9.0.log)

GregValiant, 2021-04-24:
I read the gcode files into AutoCAD (to take Cura out of the loop).  I didn't see any odd moves.  The image below shows the 4.8 brim file overlayed onto the 4.9 brim file.  The file sizes are different so they aren't identical but they are pretty darn close.  (The two green lines that go to X0 Y235 are the "present print" moves at the end of each file.)

[image]

Temperatures In both the 4.8 and 4.9 files are identical.  There are 3 M104 lines in each.  The first sets the Initial Layer Print Temperature to 230.  The second at around line 3943 lowers the temp to 220.  The final M104 turns the hot end off at the end of the gcode.  So I don't see any problems with the Gcode.  I dry ran the 4.9 file on my Ender 3 Pro from 1/2 way through the first layer and into the 2nd layer and there weren't any problems.  The temperature changed as it should.

fvrmr, 2021-04-26:
Hi @DrFailbucket thank you for your report.
Like @GregValiant I can't see anything strange that will cause this. 
So it could be that your firmware can't work with the temperature settings. Maybe you could change the temperature and see if that doesn't break the print.

DrFailbucket, 2021-04-26:
Hi, thanks for the response.

I was trying some simple shapes,Calibration Cubes, Stringing Tower and Temp Tower. All shows the same issue.
Then i Printet 10x an Calibration X what is just 1 layer tall and it printed fine. When i try to Print the X with 5layers (stacked on eachother) then it shows the also the issue. 

I'll try and run an PID tune and see if it causes the Problem.

DrFailbucket, 2021-04-26:
OK so after running the PID it Prints. 

But what i don't understand is, why is the gcode sliced with 4.8.0 running without problems even when i didnt do a PID?

GregValiant, 2021-04-26:
I don't get it either.  The two files are virtually identical.  You can tell in my previous image that something is different because the two prints end up in different places before going into "present print".  But everything referring to temperatures is near identical.
At any rate it appears to be fixed.

GregValiant, 2021-04-27:
__

patrickmollohan, 2021-04-28:
Can confirm temperature related issues with Cura 4.9 that were not present in 4.8. Mine doesn't stop printing, but does have wild temperature swings of ±10°C of specified nozzle temperature with objects sliced in Cura 4.9, compared to ±0.2°C of specified nozzle temperature with object sliced in Cura 4.8 with the same profile and objects. Although I did try a PID autotune, this did not help (or hurt). I checked all wiring (fine as I predicted) and I also have a low-ESR capacitor mod on my motherboard to help stabilise temperatures. I also noticed there is a size difference between the two sliced files (38.2MB on the one sliced by 4.8, 38.8MB on the one sliced by 4.9).

Using Cura 4.9 AppImage for Linux (Manjaro), Tevo Tornado V2 printer with a few mods, Marlin 2.0.x bugfix (4/27/2021).

GregValiant, 2021-04-29:
@patrickmollohan - If you would post that problem gcode file from 4.9 and a good one from 4.8 somebody will take a look.  Once the printer sees an M104 or an M109 the temperature is set and it remains at that set point until another M104 or M109 changes the temperature.  If there is a problem in the printer the temperature can vary, but the only way to get it to float by +/-10° using gcode is to have a boatload of M104's/M109's in the gcode.

patrickmollohan, 2021-04-29:
@GregValiant my apologies, I wasn't too sure if I should post the gcode given an STL could be reconstructed from it. 
[Sliced Files.zip](https://github.com/Ultimaker/Cura/files/6400472/Sliced.Files.zip) contains the good and bad sliced files. I successfully printed the good one last night and it came out perfectly; throughout the entire time, I only noticed a 0.4C increase (215.4C) from where it should be (215C). If I manually set the nozzle to 215C either in my printer's menu or with OctoPrint, it stays stable for the entire duration (left it on for an hour to verify).

When I tried printing the 4.9 file, layer lines were inconsistent, and some of the grill pieces had become either blobby or shifted off, and thus I stopped the print. During the bed levelling after the nozzle is heated, the nozzle temp is fine, but as soon as it begins to extrude the 4.9 file, it'll drop to around 205C, then overcorrect, undercorrect, etc repeatedly. It never finds stability. This is repeatable with all files sliced by 4.9.

I appreciate someone taking a look into this. I've done all I can to eliminate variables on my end. The only other thing I can think of is if the 4.8 profile didn't migrate properly into 4.9? But let there be no doubt: this is an issue with 4.9, not my printer.

GregValiant, 2021-04-29:
Just looking through the 4.9 gcode file the following are the Hot End Temp commands (preceded by the gcode line number):

          21 M104 S215  ;The beginning of the file and is the Initial Print Temperature.
          22 M109 S215
          23 M109 R215
    14521 M104 S215  ;This is at the start of the second layer and is the Print Temperature.
1270052 M104 S0  ;At the end of the file.

That's it.  All the temperature commands are for the Hot End at 215.  An "M109 S" is wait for heating to complete.  An "M109 R" is wait for either heat-up or cool-down to complete (whichever applies).  Typically an "M109 R" is not generated by Cura BUT it is also present in the 4.8 file.  I'm not sure what setting would trigger M109 R to be inserted in the gcode.  Maybe someone from the Cura team can address that.

patrickmollohan, 2021-04-29:
Yeah I took a look at that after you mentioned the M104/M109 part above, yet I don't understand why it repeatably fails with 4.9. I'm currently printing the next piece of the mask as we speak, again sliced with 4.8:
[image]

I don't know if plugins have anything to do with it, but the only ones I have installed are "Calibration Shapes" v1.1.1 by 5axes and "Version Upgrade 4.7 to 4.8" v1.0.0 by Ultimaker B.V. alongside the bundled ones. I've uploaded my 4.8 profile just in case it helps: [Cura Profile.zip](https://github.com/Ultimaker/Cura/files/6401969/Cura.Profile.zip)

GregValiant, 2021-04-30:
The M109 R appears to be in your default startup gcode.  It comes from your printer definition file and is present in 4.8 and 4.9.

_"But let there be no doubt: this is an issue with 4.9, not my printer."_  You sound very positive, and yet I'm not convinced that is true.  My searches and your graph are saying that the temperature is 215 and it isn't being varied by any gcode command in that file.

patrickmollohan, 2021-04-30:
That's good to know where that comes from; I appreciate the info.

Yes, I am very positive. My graph is showing a print sliced by 4.8, as I've switched back for the time being. I agree that there doesn't appear to be any temperature command changing the temp, but all files I have thus far sliced in 4.9 has this effect. That's why I even updated my printer's firmware a couple days ago to see if there was some random bug in Marlin. I inspected the wiring, ran test prints not sliced by Cura at all (all of which came out fine). All 4.8 files are perfectly fine. Same .stl files, same printer, same filament, same profile, everything the same save for the version of Cura. I cannot eliminate more variables than that. So *perhaps* there's some really odd coincidence that when I go printing the 4.9 files, I'm getting massive fluctuations and not when printing 4.8 files, in which case I apologise profusely, but I'm again printing the next piece sliced in 4.8 and it's still just as stable as what I expect it to be. I'll run the 4.9 file again once this piece is finished and update the post just for further proof, but that's about all I can do on my end. I don't wish to waste anyone's time, and I wouldn't be here if I didn't truly believe this is some issue with 4.9. This also isn't the first time this has happened with Cura; it may have been 4.6 that I had to skip due to similar problems? Don't recall.

GregValiant, 2021-04-30:
Thinking about this, there is one thing that could affect temperature and be different between versions and that might be Flow Rate.  Varying flow rates could cause the hot end to appear to fluctuate do to the cooling effect of filament passing through.  I don't know enough about the inner workings of Cura to address that (or even to know if it is really a possibility) but I thought I would throw it up here.  I think if we were discussing the Arachne version which does alter flow rate for different extrusion paths, that it would be more likely, but again, I don't really know.  Maybe @Ghostkeeper, @jellespijker, or another member of the Cura Team might have an idea.

closed 2021-05-27 by no-response[bot]