Ultimaker/Cura #9676: Cura 4,9 and mks wifi

mitofun, 2021-04-24:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Minik

### Reproduction steps

1- upload cura 4,9,0
2- set it up
3- run it
4- tried to run mks WiFi - no luck
5- uninstall mks WiFi and installed it again 
6- cannot see the ikon (button) mks WiFi in the printers settings ;( 

### Actual results

No mks WiFi button on the printers settings 

### Expected results

No button mks WiFi in the printers settings 

### Additional information

:(

fieldOfView, 2021-04-24:
This has to be fixed by the makers of that plugin. See https://github.com/Jeredian/mks-wifi-plugin/issues/148

jellespijker, 2021-04-24:
See the comment of @fieldOfView, This is a bug in the mks-wifi-plugin and we don't maintain that code. The maintainer(s) are aware of it and working on it. I will therefore close this issue.

closed 2021-04-24 by jellespijker

mitofun, 2021-04-24:
> This has to be fixed by the makers of that plugin. See  Jeredian/mks-wifi-plugin#148

got it
thank you