Ultimaker/Cura #9677: Problems loading files

Antonyvw, 2021-04-24:
### Application Version

4.9.0

### Platform

windowss 10

### Printer

Ender 3

### Reproduction steps

1. Load file
2. Restart program
3. Load file 

### Actual results

Sometimes the file will load sometimes it aborts the program. It is happening often enough and with a selection of files to warrant concern. sometimes a file that causes the abort may load after 3rd or 4th attempt, but more often than not it does not load at all.

Antony

### Expected results

The file should load. However not always doing so.

jellespijker, 2021-04-24:
We need more information, can you provide us with the log file and a project file that exhibits this behavior?

Antonyvw, 2021-04-24:
@jellespijker 

Thanks for the response. Attached are the cura.txt file (I think that's the log - correct me if its not please), and the file that failed last.  Ive sent it as I downloaded it, zipped up as it would not let me send as an STL file.
Antony

[cura.log](https://github.com/Ultimaker/Cura/files/6369907/cura.log)

[Parametric_Creeper.zip](https://github.com/Ultimaker/Cura/files/6369939/Parametric_Creeper.zip)

fvrmr, 2021-04-26:
Hi @Antonyvw thank you for your bug report. 
I think it is related to #9668 so I will close this issue and keep you updated in that thread. So we keep everything at one place.

closed 2021-04-26 by fvrmr