Ultimaker/Cura #9678: 4.9 crashes when trying to scale eraser

rakoenig, 2021-04-24:
### Application Version

4.9.0

### Platform

openSUSE 15.2

### Printer

Sovol SV01

### Reproduction steps

1. Loaded an STL
2. Placed an eraser in the model
3. Selected that eraser
4. Click on Scale in the left icon bar

### Actual results

Cura crashes immedieately. If I run Cura from a shell I get those "famous last words":

/tmp/.mount_UltimafHyM2H/usr/bin//cura.sh: Zeile 14: 18981 Ungültiger Maschinenbefehl   (Speicherabzug geschrieben) cura "$@"

So looks like "invalid instruction encountered, core dump". 

### Expected results

Able to scale the eraser. (Works with the same model in Cura 4.7.1).

fvrmr, 2021-04-26:
Hi @rakoenig thank you for your bug report.
I am not able to reproduce this. Could you share a project file?

rakoenig, 2021-04-26:
Hi @fvrmr, I guess the problem has a hardware dependency. I copied the AppImage to my notebook, excact same Linux distribution but different CPU and it works without problems. On my desktop workstation the app image crashes immediately with 

`/tmp/.mount_Ultima93qXzl/usr/bin//cura.sh: line 14:  2370 Illegal instruction     (core dumped) cura "$@"`

in the moment I try to scale anything, it doesn't need to be an erazer. 

The CPU on my crashing workstation is:

Intel(R) Xeon(R) CPU           X5670  @ 2.93GHz

The CPU on the notebook that works is 

Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz

So I really wonder what weird things are happening.

smartavionics, 2021-05-02:
See discussion (with a temporary solution) in https://github.com/Ultimaker/Cura/issues/9709.

Ghostkeeper, 2021-05-11:
Yes, we'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/9709 . This one was reported a few days earlier, but there's a lot more information in the other one so it makes sense to continue the discussion there.

closed 2021-05-11 by Ghostkeeper