Ultimaker/Cura #9681: 4.9 creates supports in the wild

rakoenig, 2021-04-24:
### Application Version

4.9.0

### Platform

openSUSE 15.2

### Printer

Sovol SV01

### Reproduction steps

1. Downloaded STL from https://cdn.thingiverse.com/assets/e9/b9/b2/da/a5/adapter_pour_telescope_v3-slider.stl

Slicing it with 0.2 and 40% infill and supports enabled. 

### Actual results

Cura 4.9.0 creates a small support "in the wild" on the right side in the front.

### Expected results

No Supports in the wild as Cura 4.7.1 did.

### Additional information

**Screenshot from 4.7.1:**

[image]

**Screenshot from 4.9.0:**

[image]

fvrmr, 2021-04-26:
Hi @rakoenig thank you for your report.
Disable `use towers` under custom and your problem should be solved.
I will close this issue, but please reach out if you have more questions!

closed 2021-04-26 by fvrmr

Ghostkeeper, 2021-05-11:
For posterity, this is an actual bug, but a duplicate of https://github.com/Ultimaker/Cura/issues/7487.