Ultimaker/Cura #9683: Confirm Overwrite Twice

TravisLlado, 2021-04-24:
### Application Version

4.9.0

### Platform

MacOS

### Printer

Any

### Reproduction steps

Whenever I choose to save a gcode file over a pre-existing gcode file, MacOS asks me to confirm whether I want to overwrite the pre-existing file and then Cura also asks whether I want to overwrite the pre-existing file. Asking for the same confirmation twice is tedious and unnecessary.

### Actual results

Twice is unnecessary

### Expected results

Once would be sufficient

KenLucke, 2021-04-24:
Yeah, very annoying, but not new.  I've even learned exactly where to position the mouse while waiting for the second approval dialog to appear so I can click "Yes"

fvrmr, 2021-04-26:
Hi @TravisLlado thank you for your bug report. 
This is like @KenLucke mentioned indeed a known bug.
See https://github.com/Ultimaker/Cura/issues/7937 that's why I will close this issue.

closed 2021-04-26 by fvrmr

aabmar, 2022-01-24:
Still present in 4.13.0. I'm using Monterey. The double replace will appear on all save operations, when you choose to overwrite a file. The first dialog is the OS native dialog, the second is a QT dialog.