Ultimaker/Cura #9684: Tool Tip Time to Appearance Setting

KenLucke, 2021-04-24:
### Is your feature request related to a problem?

The new expanded tooltips when the Settings Guide is installed are AWESOME (and I notice that you have not added that to the list of enhancements to Cura that is being promoted - you should mention that tooltips are now enhanced when SG is installed), rivaling the Settings Guide for information, but without having to manually access that guide with a right click or the menu.

HOWEVER, they appear much too rapidly ( ≤ 1 second) when mousing around, and are often very large on screen, obscuring what you're actually working on when you're not intentionally wanting them.  

### Describe the solution you'd like

A longer delay for the tooltips to appear when mousing over a setting name, preferably with a user-configurable time setting of "X" (in Preferences > General), where "X" = number of seconds before they appear, and "0" disables them.

This would also be good for when the Settings Guide is not installed and there are just the normal, non-enhanced tooltips.  I want to consciously instigate a tooltip's appearance by hovering, not just accidentally lingering a split second too long

### Describe alternatives you've considered

Live with it the way it is

### Affected users and/or printers

Everyone.

fvrmr, 2021-04-26:
Hi @KenLucke thank you for your feature request.
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-04-28:
This was discussed among the Cura team but we've decided to reject it.

A preference to determine the delay of setting tooltips is such a tiny detail that we think it only muddles the list of preferences and almost nobody would care about that preference.

We could look into creating the defaults, but the current delay seems to be fine for the normal Cura behaviour (without Settings Guide, that is).

The expanded tooltips have been part of the Settings Guide for about a year now (first released with [version 2.2](https://github.com/Ghostkeeper/SettingsGuide/releases/tag/v2.2.0) of the Guide). But with this release I've set it to be enabled by default, because it doesn't do a lot of harm and helps people find the guide. I've had some people even that were giving me suggestions for things to add to the guide by pointing at the original Cura tooltips and saying that things were missing; not knowing that the actual guide wasn't even being shown to them. Cura itself has nothing to do with the expanded tooltips, as the change is kind of [hacking into Cura's internals](https://github.com/Ghostkeeper/SettingsGuide/blob/75a902269e6d1598dce0add486447854657cd559/CuraSettingsGuide.py#L121-L124). Cura did have a feature in its change log a while back that tooltips are now scrollable, which was implemented specifically to facilitate this feature of the Settings Guide. But it's really just a feature of the Settings Guide, not of Cura.
I could look into adding a preference to the Settings Guide to widen the tooltips or not. But for Cura this issue is rejected since the delay of the tooltips is not seen as a preference that anyone will really care about to change.

closed 2021-04-28 by Ghostkeeper