Ultimaker/Cura #9685: Cura 4.9 software crash, Linux Mint 64 bit

jk101290, 2021-04-25:
### Application Version

4.9

### Platform

Linux Mint

### Printer

Ender 3 Pro

### Reproduction steps

1. Import an STL
2. Click on the model and then the "rotate" button on the left side of the screen 

### Actual results

Cura software completely closes/exits

### Expected results

After clicking the rotate button, Cura should allow me to rotate my selected model.

### Additional information

This occurs for me on the Cura 4.9 appimage for Linux 64 bit. After importing an STL, I select my model, select "rotate" and the software crashes/exits. It occurs every time I try to use Cura 4.9. Everything works normally in Cura 4.8 appimage.

fieldOfView, 2021-04-25:
Could you please post a link to your cura.log?
https://github.com/ultimaker/cura/#logging-issues

jk101290, 2021-04-25:
> Could you please post a link to your cura.log?
> https://github.com/ultimaker/cura/#logging-issues

https://drive.google.com/file/d/1oN3feChVm7MHv8kw3gapibFMCw3VAcuY/view?usp=sharing

Ghostkeeper, 2021-05-12:
Hmm, in that log file it looks like the rotate operation was successful, and you started a slice afterwards. Here is a relevant snippet.

```
2021-04-25 09:07:26,955 - DEBUG - [JobQueWorker [0]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.08351349830627441 seconds. 2258 input vertices. 2258 output vertices.
2021-04-25 09:07:26,970 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0xe97dc00>, took 1ms
2021-04-25 09:07:26,994 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.0,0.0,0.0,w=4.0>) TranslateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,trans.=<0.0,0.0,0.0>), took 7ms
2021-04-25 09:07:28,702 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 10ms
2021-04-25 09:07:28,805 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.49999999999999994,0.0,0.0,w=0.8660254037844387>), took 6ms
2021-04-25 09:07:28,911 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 6ms
2021-04-25 09:07:29,047 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 8ms
2021-04-25 09:07:29,181 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 12ms
2021-04-25 09:07:29,244 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,1.2,0.0>), took 2ms
2021-04-25 09:07:30,936 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 6ms
2021-04-25 09:07:31,047 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 8ms
2021-04-25 09:07:31,242 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,1.16,0.0>), took 4ms
2021-04-25 09:07:32,300 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 25ms
2021-04-25 09:07:32,650 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,-0.759,0.0>), took 17ms
2021-04-25 09:07:33,978 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 14ms
2021-04-25 09:07:34,317 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,-1.6,0.0>), took 2ms
2021-04-25 09:07:37,309 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-25 09:07:41,453 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 17ms
2021-04-25 09:07:41,579 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.25881904510252074,0.0,0.0,w=0.9659258262890683>), took 14ms
2021-04-25 09:07:41,678 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 6ms
2021-04-25 09:07:41,981 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 21ms
2021-04-25 09:07:42,159 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Back_bumper.STL'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 24ms
2021-04-25 09:07:42,566 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,-3.18,0.0>), took 2ms
2021-04-25 09:07:47,661 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [256]: Creating thumbnail image (just before slice)...
2021-04-25 09:07:47,685 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2021-04-25 09:07:47,701 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultima9fsKkA/usr/bin/resources/shaders/overhang.shader]...
2021-04-25 09:07:47,709 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultima9fsKkA/usr/bin/resources/shaders/transparent_object.shader]...
2021-04-25 09:07:47,715 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_Ultima9fsKkA/usr/bin/resources/shaders/striped.shader]...
[... cut ...] n Cura 4.9 (as shown earlier during the loading of the plug-ins including the 48to49 upgrade).

Did it crash after the slice started? From your original post I'd have gathered that it would happen upon clicking the Rotate tool.

In either case, I haven't been able to reproduce this issue on Ubuntu 20.10, which should be very similar to your Mint. It's either something in the reproduction steps (unlikely since they are very simple) or something different about the environment that makes it crash.

jk101290, 2021-05-12:
The issue seems to have gone away in the release of 4.9.1, thanks!

closed 2021-05-12 by jk101290