Ultimaker/Cura #9687: Cura 4.9 FLSUN QQ-S view has wronk orientation

berberic2, 2021-04-25:
### Application Version

4.9.0

### Platform

Linux

### Printer

FLSUn QQ-S Pro

### Reproduction steps

Simply start Cura with QQ-S profile. The model shown has wrong orientation
[image]
left: Cura 4.8 correct
right: Cura 4.9 wrong

### Actual results

Wrong orientation

### Expected results

Corect orientation

fvrmr, 2021-04-26:
Hi @berberic2 thank you for your report.
We don't maintain these printers. It seems like the meshes are updated. You could copy the mesh from 4.8 and paste it into the 4.9 meshes folder and change the printer def for the flsun qq-s as a work around. If you have a fix yourself a pull request is always welcome. 
I will close this issue since we will not fix this. But please reach out if you need help or have more questions.

closed 2021-04-26 by fvrmr

berberic2, 2021-07-11:
Problem persists in 4.10

fvrmr, 2021-07-13:
Hi @berberic2 thank you for letting me know. But we still don't maintain this, you can change it manually like I mentioned before or use this workaround: https://github.com/Ultimaker/Cura/issues/9738#issuecomment-830695645 but you have to do this yourself for every release if it is not fixed by a pull request. 

Here you can read how to create a pull request for this: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 
Let me know if you need help with that.

berberic2, 2021-09-27:
The problem was introduced by this commit: #https://github.com/Ultimaker/Cura/commit/9ffc6bc70587c5e53a43fba47f3aa1bc20eda87d

the new position is wrong.

berberic2, 2021-12-04:
Still wrong in 4.12.1

skoroneos, 2022-03-28:
Greetings.
I took sometime today and fixed the flsun_qq_s.3mf file and now displays correctly in 4.13.1
I am attaching the file here for others to test also and if its OK, ask someone from the devs to push it into the next release
[flsun_qq_s.zip](https://github.com/Ultimaker/Cura/files/8365816/flsun_qq_s.zip)

berberic2, 2022-04-24:
Looks O.K. for me.
But considering the history of this bug (#9687  #9738 #11544 #8663) I doubt it will be fixed.
The mantainers seems to trust @LuizPonzini, which has filed exactly one bug ever (this one), more.

skoroneos, 2022-04-29:
Thanks guys for taking time to look into this

RVillani, 2022-04-30:
> Thanks guys for taking time to look into this

I don't work with them. I'm just another user who was very happy with the fixed model you provided.
They made it clear they don't maintain the third party printer models, but seem open to pull requests.
I hope they merge the pull request so that we can have this fixed version of the model by default.

commit 9fea0b2f (2022-07-19): Fix #9687: FLSUN QQ-S preview mesh rotated to face -Y

Thanks to @skoroneos, who provided the file with the correct orientation.
PR #11968 (2022-07-19): Fix #9687: FLSUN QQ-S preview mesh rotated to face -Y
The viewport preview mesh for the *FLSUN QQ-S* printer now faces the correct direction, which is -Y.
Thanks to @skoroneos, who provided the file with the correct orientation. I tested it and it perfectly matches what I see on the printer while it's printing.

I'm requesting this pull to fix #9687 as suggested by @fvrmr on the issue comments. This seems to be the correct repository to request the pull for, as I didn't find the path `resources/meshes` on the other repositories.