Ultimaker/Cura #9689: Unexpected termination after slicing and clicking preview, closes CURA or crashes IMAC

ReiDalla, 2021-04-25:
### Application Version

4.9.0

### Platform

MACOS High Sierra 10.13.16

### Printer

Anycubic MEGA X

### Reproduction steps

After slicing any STL and when clicking on preview, the CURA locks or even the IMAC locks in such a way that just using the reset button to go back.

### Actual results

After slicing any STL and when clicking on preview, the CURA locks or even the IMAC locks in such a way that just using the reset button to go back.

### Expected results

Solve

### Additional information

Process:               cura [1303]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.9 (4.9.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [1303]
User ID:               501

Date/Time:             2021-04-25 10:41:59.740 -0300
OS Version:            Mac OS X 10.13.6 (17G14042)
Report Version:        12
Anonymous UUID:        40FB058E-7036-1D74-3BD0-F3264A5BD968

Time Awake Since Boot: 1200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000062b40703
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x62b40703:
--> 
    __TEXT                 0000000106258000-00000001064a0000 [ 2336K] r-x/r-x SM=COW  /Applications/Ultimaker Cura.app/Contents/MacOS/cura

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff60853b66 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff60a1e080 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff607616fe raise + 26
3   libsystem_platform.dylib      	0x00007fff60a11f5a _sigtramp + 26
4   ???                           	000000000000000000 0 + 0
5   libQt5Qml.5.dylib             	0x0000000118f7ccfc QV4::QObjectWrapper::getProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*) + 2924
6   libQt5Qml.5.dylib             	0x0000000118f7d794 QV4::QObjectWrapper::getQmlProperty(QV4::ExecutionEngine*, QQmlContextData*, QObject*, QV4::String*, QV4::QObjectWrapper::RevisionMode, bool*, QQmlPropertyData**) + 436
7   libQt5Qml.5.dylib             	0x0000000118f61c44 QV4::QQmlContextWrapper::lookupInParentContextHierarchy(QV4::Lookup*, QV4::ExecutionEngine*, QV4::Value*) + 436
8   ???                           	0x0000000130fbca99 0 + 5116775065
9   libQt5Qml.5.dylib             	0x0000000118f9bf93 0x118e74000 + 1212307
10  libQt5Qml.5.dylib             	0x0000000118f408c6 QV4::Function::call(QV4::Value const*, QV4::Value const*, int, QV4::ExecutionContext const*) + 342
11  libQt5Qml.5.dylib             	0x00000001190c4273 QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) + 579
12  libQt5Qml.5.dylib             	0x00000001190cbf3d 0x118e74000 + 2457405
13  libQt5Qml.5.dylib             	0x00000001190ca050 QQmlBinding::update(QFlags<QQmlPropertyData::WriteFlag>) + 304
14  libQt5Qml.5.dylib             	0x00000001190ab1d9 QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 713
[... 964 lines cut ...]
TOTAL, minus reserved VM space     2.1G     2251 

Model: iMac11,3, BootROM 99.0.0.0.0, 2 processors, Intel Core i3, 3,2 GHz, 12 GB, SMC 1.59f2
Graphics: ATI Radeon HD 5670, ATI Radeon HD 5670, PCIe
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x830B, 0x4E54344743363442384842304E532D434720
Memory Module: BANK 0/DIMM1, 4 GB, DDR3, 1333 MHz, 0x830B, 0x4E54344743363442384842304E532D434720
Memory Module: BANK 1/DIMM1, 4 GB, DDR3, 1333 MHz, 0x830B, 0x4E54344743363442384842304E532D434720
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x8F), Atheros 9280: 4.0.74.0-P2P
Bluetooth: Version 6.0.7f22, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: WDC WDS100T2B0A-00SM50, 1 TB
Serial ATA Device: OPTIARC DVD RW AD-5680H
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB Optical Mouse
USB Device: Internal Memory Card Reader
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB Reader
USB Device: USB Keyboard
USB Device: IR Receiver
USB Device: Built-in iSight
Thunderbolt Bus:

whughes98144, 2021-04-26:
I've the same problem with same repro.  Reverting to previous version fixed it.

fvrmr, 2021-04-26:
Hi @ReiDalla thank you for your bug report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9631 and that's why I will close this issue.

closed 2021-04-26 by fvrmr