Ultimaker/Cura #9691: Unable to slice grouped models

BagelOrb, 2021-04-25:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. load in a couple of cubes in the second extruder
2. disable first extruder
3. reorient them
4. group them
5. reorient them

Load in 

### Actual results

[image]

And then when I saved the 3mf file there were suddenly extra greyed out areas:

### Expected results

Slicing

### Additional information

Logs say something weird about `Cannot read property 'horizontalCenter' of null` :

```
2021-04-25 16:33:47,887 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [271]: Starting to slice...
2021-04-25 16:33:47,888 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [290]: Going to slice build plate [0]!
2021-04-25 16:34:17,922 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null
2021-04-25 16:35:49,374 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [135]: Writing to [E:/Downloads/UM3_cube_1.3mf.renamed.zip]...
2021-04-25 16:35:49,378 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [100]: There was no auth data or access token
2021-04-25 16:35:49,387 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [156]: Writing to Local File E:/Downloads/UM3_cube_1.3mf.renamed.zip in binary mode
2021-04-25 16:35:49,478 - WARNING - [MainThread] UM.Settings.InstanceContainer.setCachedValues [161]: Unable set values to be lazy loaded when values are already loaded 
2021-04-25 16:35:50,228 - WARNING - [MainThread] UM.Settings.InstanceContainer.setCachedValues [161]: Unable set values to be lazy loaded when values are already loaded 
2021-04-25 16:35:50,585 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition ultimaker2_plus failed to pickle.
2021-04-25 16:35:50,901 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition ultimaker2_extended_plus failed to pickle.
2021-04-25 16:35:51,228 - DEBUG - [JobQueWorker [2]] UM.FileHandler.WriteFileJob.run [80]: Writing file took 1.8319971561431885 seconds
2021-04-25 16:35:51,258 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null
2021-04-25 16:35:51,351 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-04-25 16:35:51,357 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-04-25 16:35:51,918 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-04-25 16:36:01,246 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-25 16:36:21,250 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null

```

fieldOfView, 2021-04-25:
Are you using the Sidebar GUI? (Cool!). The message about horizontalCenter of null is a message from QML, and will have nothing to do with the orientation or groupings of models, but only with the placement of the MessageStack on the screen. I would say that is unrelated, but I'll have a look at it anyway.

fieldOfView, 2021-04-25:
A project file would be nice to see what is going on (but you knew that, right?)

BagelOrb, 2021-04-25:
When filling in the template forms the uploading on github doesn't seem to work correctly. Let me reupload.

fvrmr, 2021-04-26:
Hi @BagelOrb it seems like it is related to this disabling the first extruder, then it is a duplicate of this: https://github.com/Ultimaker/Cura/issues/9322

BagelOrb, 2021-04-26:
Seems like it.

closed 2021-04-26 by BagelOrb