Ultimaker/Cura #9692: Cura 4.9.0 stürzt beim laden eines Bauteiles (.stl) direkt ab. Cura crashed while loading stl-binary-file

SirWinni, 2021-04-25:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Load Cura 4.9.0
2. Load .stl-file
3. Cura crash befor file beginn loading

### Actual results

Cura is crashrd

### Expected results

cura crashed

### Additional information

a screenshoot is inpossible to make. Cura crashed and cura-Window closed.

nallath, 2021-04-25:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

SirWinni, 2021-04-25:
Prima, ich weiß nicht was sie von mir wollen. Im Text steht alle drin. Muss man bei euch englisch können um einen Fehler zu melden.???
Danke für die Sturheit. Schlimmer als deutsche Paragraphenreiter. Und danke für die nicht erfolgte Hilfe. Nie wieder Ultimaker!!!!

Mit freundlichen Grüßen
Winfried Schmickler

nallath, 2021-04-26:
My German is a bit rusty, so sorry if I didn't fully understand what you wrote. My German is about enough to get by on a holiday, but it's nowhere near enough to provide technical support, sorry.  

You're missing a log file, as is mentioned in both the template and the article that I linked you. We really can't help you if you don't provide us with that.

Siriusbizzy, 2021-04-30:
I have the same thing happening to me with CURA 4.9.0, however I'm on Windows 7 x64, with a Creality CR10 S5.
It's a crash to desktop every time as soon as I go to open a file and then (try to) select an STL file, the application just completely vanishes.
I'll attach my log file, but let me know if you want me to open a separate case.
[cura.log](https://github.com/Ultimaker/Cura/files/6408439/cura.log)

fieldOfView, 2021-05-01:
@Siriusbizzy try this on Windows 7 x64:
https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-679942344

The above does not apply to Windows 10 as far as I know, so it will not help @SirWinni

Siriusbizzy, 2021-05-01:
> @Siriusbizzy try this on Windows 7 x64:
> [pyinstaller/pyinstaller#4855 (comment)](https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-679942344)
> 
> The above does not apply to Windows 10 as far as I know, so it will not help @SirWinni

Well, @SirWinni ragequit the forum if you read the German (which I did), so I just want to save the topic :)

fieldOfView, 2021-05-01:
So, did you try installing [KB3126587](https://www.catalog.update.microsoft.com/Search.aspx?q=KB3126587%20windows%207) as suggested in the issue I linked above?

Siriusbizzy, 2021-05-02:
> So, did you try installing [KB3126587](https://www.catalog.update.microsoft.com/Search.aspx?q=KB3126587%20windows%207) as suggested in the issue I linked above?

I did and unfortunately it was already installed, the other update mentioned in that thread was not applicable to my system.
Not sure what other info could be provided to solve this one.

Ghostkeeper, 2021-05-12:
Your log file indicates there's a problem with the striped shader:

```
2021-05-01 00:05:38,257 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setFragmentShader [139]: Fragment shader failed to compile: 0:14(34): error: syntax error, unexpected ';', expecting '{'
```

In your log file it indicated that you're using the compatibility mode (because OpenGL 4 is not supported yet by your graphics driver). And it indicates a line number. Looking at recent changes, it seems quite obvious that we have a mistake here: https://github.com/Ultimaker/Cura/commit/3c4d0b6ecdecfd92ffbf7954d1fc3b18ee21586a

The omission of the type there is inconsistent with the OpenGL4 version and also a recent-ish change (new in 4.9). It shouldn't theoretically cause it to fail compiling as it should default to int, but int is the wrong type there anyway, and perhaps some graphics drivers don't have a lowp int or just didn't implement the default.

commit 3c4d0b6e (2021-05-12): Add missing type to lowestPrintableHeight variable

Otherwise it should theoretically default to int, which is wrong. However with some graphics drivers it just seems to result in a failure to compile and concluding with a crash and crescendo.

Fixes #9692.
closed 2021-05-12 by Ghostkeeper