Ultimaker/Cura #9694: Fan speed incorrect

rwetzel67, 2021-04-25:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Anycubic I3 Mega

### Reproduction steps

I set the fan speed as below:
Enable cooling ON
Fan speed 60%
Initial fan speed 0
Regular fan speed at layer 3
[image]

### Actual results

Fan speed in GCODE (command M106) is set to 100% at layer 3 (50% at layer 2).
I tried different fan speeds (10%, 20%) , same result (always get M107 before layer 0; M106 127.5 before layer 1 and M106 255 before layer 2)

### Expected results

for fan speed 60% I expect
M107 before layer 0
M106 76,5 before layer 1
M106 153 before layer 2

fvrmr, 2021-04-26:
Hi @rwetzel67 thank you for your bug report.
I could reproduce this with a small model. When I scaled the model bigger the expected results happen. 
If you want you could share a project file so I can take a closer look.

rwetzel67, 2021-04-26:
It is indeed a small model, around 30 x 30 mm; 30 min print at 40mm/s. I can share the project by the end of the week, I am not at home now. I did not check this in other models, will do it later.

rwetzel67, 2021-04-29:
Attached project. I checked again and issue is present. I could not send the file here, pls download it from Google Drive
https://drive.google.com/drive/folders/1kMVHY5V2t52VAY2IZ7YUTmNd1OiFG-Mf?usp=sharing

Ghostkeeper, 2021-05-12:
It is indeed the minimum layer time.

In your project file if I filter the g-code on `;LAYER|M106|M107` I get the following lines:

```
M107 ;start with the fan off
;LAYER_COUNT:164
;LAYER:0
M107
;LAYER:1
M106 S127.5
;LAYER:2
M106 S255
```
So the fan speed is set to 0 for the first layer through the `M107` command (because of the Initial Layer Fan Speed). Then towards layer 3 it grows to the Maximum Fan Speed (which is set to 100%) rather than the regular fan speed.

If you reduce the Minimum Layer Time and Regular/Maximum Fan Speed Threshold to 0, you get this result instead:

```
M107 ;start with the fan off
;LAYER_COUNT:164
;LAYER:0
M107
;LAYER:1
M106 S76.5
;LAYER:2
M106 S153
```

255*0.6 = 153, and 255*0.3 = 76.5, so that result seems to be correct then.

closed 2021-05-12 by Ghostkeeper

rwetzel67, 2021-05-12:
Ok, thanks. I'll check the minimum layer time and max fan speed in my configuration.