Ultimaker/Cura #9695: Parameters menu jumps when after search and a selection

mbourque, 2021-04-25:
### Application Version

4.9.0

### Platform

OSX

### Printer

Creality CR-6 SE

### Reproduction steps

1. Open Cura
2. Show all Settings
3. Search for 'Support'
4. Scroll down for the checkbox to enable support
5. Click to enable support 

### Actual results

The settings menu jumps out of place requiring you to scroll and find it again. It's a pain and bad UX

### Expected results

Would prefer it to stay put so I can change the other parameters related to support.

mbourque, 2021-04-25:
[2021-04-25_08-30-23.mp4.zip](https://github.com/Ultimaker/Cura/files/6372414/2021-04-25_08-30-23.mp4.zip)

fvrmr, 2021-04-26:
Hi @mbourque thank you for your bug report.
I can reproduce this and will discuss this with the team. Keep you posted.

mbourque, 2021-04-26:
Thanks, it is very annoying and makes using the product hard to work with.

Ghostkeeper, 2021-04-28:
We've discussed this issue in the team and with our UX engineer. While it's a problem we've known about for years (especially with support), we don't see it as a big issue.

Our preferred solution would be that it changes the scroll position such that the setting you most recently changed remains in the same place. That would be nice.

However this is deemed to be fairly difficult, so we've decided to defer it. It's a slight problem, but not worth the effort vs. gain here to warrant giving this any priority over other issues.

mbourque, 2021-05-03:
OK - I understand and glad you know about it. I manage a UX team so this stuff bugs me.

closed 2026-09-27 by mbourque