Ultimaker/Cura #9696: Allow prime/wipe tower for single extruder

spoogemcfuck, 2021-04-25:
### Is your feature request related to a problem?

I'm doing time lapses where the print head moves out of the way for each snapshot, and I'm having issues with oozing during the snapshot. When the snapshot is complete, the head moves back and transfers the ooze to the print.

Placing a priming tower near the spot where the head parks eliminates this problem, as the ooze is wiped on the tower and the nozzle returns clean.

Unfortunately Cura doesn't allow me to turn on its priming tower settings for a single-extruder print. The settings do not appear at all in the "Print settings" list, so there's no way to turn it on.

### Describe the solution you'd like

Show the priming tower settings even in single-extruder mode.

Perhaps there's a technical reason it doesn't work for a single extruder, but if not then it would be very convenient to have Cura generate a priming tower.

### Describe alternatives you've considered

I can currently add a separate model to use as a priming tower, position, size it, worry about bed adhesion etc.

It works but it's a small hassle, and I suspect that Cura already has the capability to generate one that does a better job for less material.

### Affected users and/or printers

Anyone who does time lapses or any other situation where the print head moves off the print at regular intervals and has an opportunity to ooze.

Ghostkeeper, 2021-04-28:
We've discussed this with the team. We recognise that the oozing after the screenshot could be a problem. There is no technical reason why the prime tower couldn't be printed in a single extrusion print. It is currently limited in height until the height of the last extruder switch but that could be changed.

However printing the prime tower with a single extrusion print may not be the right solution here. Wiping in the infill is perhaps more appropriate than printing an entire tower.

More to the point though, this is not a problem that we can spend Ultimaker's time on. Ultimaker's printers don't support the screenshot command at all. For this reason we've decided to defer this issue. The oozing is a real problem, but a fix would have to come from the community (which gave us the screenshot script in the first place). Perhaps the script itself could simply be adjusted to reduce or fix the problem.

spoogemcfuck, 2021-04-28:
Thank you for taking a look anyway.

closed 2023-07-13 by github-actions[bot]