Ultimaker/Cura #9699: F11 Full Screen

Z3DPrintZ, 2021-04-26:
### Application Version

4.8.0

### Platform

Window 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

F11 Full Screen

The tabs at the top of the screen would not allow me to view the drop down menus, also the screen would flash when I clicked on each tab. 

### Actual results

Screen flashes when trying to select the tabs at the top and does not allow the drop down menus to appear.

### Expected results

Just that.

fvrmr, 2021-04-26:
Hi @Z3DPrintZ thank you for your report.
I am not able to reproduce this. Could you share your log file?

fieldOfView, 2021-04-26:
I can reproduce this, on Windows 10.

### Reproduction steps:

* Launch Cura 4.9
* Press F11 to go full screen
* Try to access the File, Edit or any other menu with the mouse
* Try to access the File menu with Alt-F, then press O (shortcuts for File -> Open...)

### Actual results:

When trying to access the File, Edit etc menus with the mouse, the menus do not open
When accessing the menus by their keyboard shortcuts, the menos don't show but the options can be selected.

# Additional info:

The context menu of the viewport is also not accessible (pressing the right mouse button over the viewport).
I tried with and without the SidebarGUI.

Z3DPrintZ, 2021-04-27:
Not sure what a log file is but, thank you @fieldOfView. I am new to Cura and just want to help when I can.

fieldOfView, 2021-04-27:
For reference, the log file is a file named `cura.log`, which can be found in the configuration folder. See `Help -> Show configuration folder...`.

fvrmr, 2021-04-29:
Thank you @fieldOfView for your reproduction steps. Tried it with Windows 10 but I can access the menu's with the mouse and shortcuts.
So a log file would be helpful.

fieldOfView, 2021-04-29:
Here's a cura.log, starting from an empty configuration folder. After adding a new non-networked printer, I went to fullscreen (using F11), then tried to access the File, Edit, View, Extensions & Help menu by mouse (which highlighted the menu name, but did not show the menu), followed by right-clicking the viewport (which did not show the context menu). Then I quit Cura.

[cura.log](https://github.com/Ultimaker/Cura/files/6399747/cura.log)

fvrmr, 2021-05-07:
I have discussed this issue with the team. And we had a ticket for this a long time on our backlog but it was taken off because of it isn't a priority for us. So I will defer this issue for now. 
Hope you understand.

closed 2023-06-09 by github-actions[bot]