Ultimaker/Cura #9700: Crash when opening any STL file

LouDnl, 2021-04-26:
### Application Version

Cura 4.9.0

### Platform

Windows 10 64 bit

### Printer

Ender 3 v2

### Reproduction steps

Install Cura 4.9.0, start Cura and open any STL file through file menu or drag STL into cura.

### Actual results

Cura crashes.

`Traceback (most recent call last):
  File "C:\Users\thier\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\thier\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'
`

`Thread 0x00003c20 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000ddc (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002478 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002f2c (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000104c (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000038bc (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000031f4 (most recent call first):
  File "threading.py", line 302 in wait
[... 96 lines cut ...]
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
`

### Expected results

Cura should open the STL file

### Additional information

**Good to note is that I still have Cura 4.8.0 installed**

LouDnl, 2021-04-26:
[cura.log](https://github.com/Ultimaker/Cura/files/6374862/cura.log)
[image]

LouDnl, 2021-04-26:
[stderr.log](https://github.com/Ultimaker/Cura/files/6374868/stderr.log)

nallath, 2021-04-26:
Please update your plugins. The custom naming plugin has had an update that fixes this.

closed 2021-04-26 by nallath