Ultimaker/Cura #9705: New Installed Materials not showing up in the list

jcastro, 2021-04-26:
### Application Version

4.9.0

### Platform

macOS Big Sur

### Printer

Artillery Genius

### Reproduction steps

1. Install new Material, in my case Smartfil PLA
2. Material should appear in the Material list to select

### Actual results

Material is not showing up in the list

### Expected results

Material should be selectable from the Material list

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No additional info, but attaching configuration files
[Uploading Archive.zip…]()

fvrmr, 2021-04-29:
Hi @jcastro thank you for your report.
I can't download you configuration files. Could you try uploading them again?

jcastro, 2021-04-29:
Sure!

[Archive.zip](https://github.com/Ultimaker/Cura/files/6396876/Archive.zip)

fvrmr, 2021-04-29:
Hi @jcastro it looks like the materials you are installing is 2.85 mm and your printer uses 1.75 mm. So that's why you can't see them.

jcastro, 2021-04-29:
@fvrmr I'm installing this one for example https://marketplace.ultimaker.com/app/cura/materials/Smartmaterials3d/SMARTFIL_PLA_GLITTER but there's no option to choose between 2.85 or 1.75mm. The filament is definitively available in 1.75 as it's the one I have

jcastro, 2021-04-29:
Also! I'm trying to isntall a lot of different brands and I can't see any of them. If there's an option to install the 1.75mm version I can't find it

fvrmr, 2021-04-29:
Looking at the material files, it show that the diameter is 2.85mm so it is not compatible with 1.75mm. That's the reason why you can't see them.

jcastro, 2021-04-29:
@fvrmr would it possible to tell me one brand that has 1.75 profile so I can install to test if it shows up? I've tried a lot of different brands and none of them showed up. thanks!

fvrmr, 2021-05-07:
Materials in the marketplace are for Ultimaker printers, and they don't have 1.75 mm profile.

closed 2021-05-07 by jcastro

Ghostkeeper, 2021-05-13:
The underlying problem is that the Marketplace doesn't show which materials are compatible with your current printer (or for which printers they are compatible).

I'd consider this a duplicate of https://github.com/Ultimaker/Cura/issues/6236