Ultimaker/Cura #9706: Cura 4.9 Preview very slow and hanging

GrooverFromHolland, 2021-04-26:
### Application Version

4.9.0

### Platform

Windows 10 Pro build 21364.1000 , NVIDIA Quadro RTX 3000

### Printer

Creality Ender 5 plus

### Reproduction steps

Preview Sliced STL.

### Actual results

In preview: Layer selection slider not reacting for several seconds, animation brim corner speed is 0.2 mm per second. 
Re-slicing file takes less than 2 seconds, so only problems in preview.

### Expected results

In version 4.8 and Aragne no problems

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Uploaded files:

jellespijker, 2021-04-27:
It seems that something went wrong with uploading log and project files. Can you added them again.

GrooverFromHolland, 2021-04-27:
Trying again to upload.
[cura.log](https://github.com/Ultimaker/Cura/files/6382834/cura.log)
[CE5P_PedalBx.zip](https://github.com/Ultimaker/Cura/files/6382837/CE5P_PedalBx.zip)

Ghostkeeper, 2021-05-13:
I'm unable to reproduce this problem so far. The layer view slider reacts pretty much immediately after the slicing is completed.

From your log file I can see that happen though. This is a relevant snippet of the log:

```
2021-04-26 23:26:14,230 - DEBUG - [JobQueWorker [0]] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.22499442100524902 seconds
2021-04-26 23:26:15,986 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [956]: See if there is more to slice(2)...
```

Notice how there's 1.75s between these two log entries. This is how that looks on my computer:

```
2021-05-13 21:11:11,061 - DEBUG - [JobQueueWorker [11]] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.36545538902282715 seconds
2021-05-13 21:11:11,086 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [956]: See if there is more to slice(2)...
```

For me there is 0.025s between the two log entries. So what is it doing in that period? The `Processing layers took X seconds` message is the very last thing of the `ProcessSlicedLayersJob`. The job has a `finished` signal which triggers the `_onProcessLayersFinished` function. Before logging that second line, that function:

* deletes the layer data received from the engine to free up memory (it's stored as a 3D mesh then), and
* deletes the `ProcessSlicedLayersJob` instance.

So the delay is either from processing the signal notification list (which seems unlikely because then your whole application would be slow), or it's from freeing up that memory. Freeing up memory should be pretty much instant. It is pretty much instant for me. I don't see how it could take that long.

GrooverFromHolland, 2021-05-14:
Layer view  compatibility mode does not change anything. Moving camera in layer view animation takes about one second to response. Also  all all mouse clicks have a delay.
Closing the program make Cura unresponsive and i have to force it to close.
I looked in Windows Resource monitor to find out more.
GPU usage is 0.5%
[image]
[image]

Average CPU use is 29 % overall and Cura about 6%
Cura.exe threads = 73
Analyze Wait Chain gives a message: One or more threads of Cura.exe are waiting for Network I/O.
I added a screenshot  of Analyze Wait Chain.
Is more information needed?

GrooverFromHolland, 2021-06-18:
Update to Ultimaker Cura 4.10.0 Beta solved my problem with the preview and is now also more responsive to mouse clicks , going to do a test print tomorrow to check if all other things are OK.

What has changed in Cura 4.10, that fixed this?

GrooverFromHolland, 2021-06-21:
No!  Update to Ultimaker Cura 4.10.0 did not do it! For all the versions 4.9, 4.9.1 and 4.10 a fresh install solved the problem for first use. Opening a second time the problem is back!

fieldOfView, 2021-06-21:
Does unchecking the option to "Restore window position on start" (and restarting Cura) make any difference?

GrooverFromHolland, 2021-06-21:
No difference.

Ghostkeeper, 2021-06-25:
This is indeed not solved in 4.10 as far as I know since none of us have been able to reproduce the issue so far.

GrooverFromHolland, 2021-06-25:
I do not have  PowerToys installed.

jellespijker, 2021-06-26:
@fieldOfView I personally have PowerToys installed on my Windows partition specifically for the FancyZones and I do not notice any lag.

fieldOfView, 2021-06-27:
@jellespijker: Then you probably don't have the SidebarGUI plugin installed.

nallath, 2021-06-28:
The restore window thing should have been fixed for 4.10 (We had a number of reports that indicate that it was indeed solved)

GrooverFromHolland, 2021-06-28:
I do not know what you mean by the restore window thing. It is not related to my problem, I think.
 I have trouble with my prepare and preview window! And I have no plugins installed myself.

But….., I am making progress. If I disconnect from the internet and then start Cura 4.9.0, the load time of the of the program is drastically improved. Loading time is reduced, from sometimes longer than a minute, to 18 seconds. Also, the problems with the prepare and preview windows are almost gone. Reactions to mouse-clicks and dragging sliders are almost instantaneous, but not as fast as Cura 4.8 and Aragne. 
This must give you a clue of what is changed since 4.8 and why it could affect my system and not many others.

Ghostkeeper, 2021-06-28:
Could you try this in Cura 4.9.1, rather than 4.9.0? One of the bug fixes in 4.9.1 was that some Python modules were trying to load from the `X:` drive, since this was the drive being used on our build server when the file paths got frozen into the cura.exe package.

GrooverFromHolland, 2021-06-28:
Uninstalled 4.9, installed 4.9.1 with no extra features, just standard installation.
With internet connected:
First time Loading with internet connection 3 minutes.
Loading my .stl takes some time, but all is OK.
Second time Loading with internet 40 seconds.
All problems are back!
Third time Loading NO internet 30 seconds.
Loading my .stl takes some time.
No difference, so still problems.
Installed 4.9.0 with 4.9.1 still installed:
First time 4.9.0 with internet no problems.
Second time 4.9.0. All problems are back!
Internet connection makes no difference anymore.
Uninstalling all versions except Ultimaker Cura Arachne_engine_alpha 99.9.0, removing all files from \AppData\Roaming\cura  and reinstalling 4.9.0. restart my computer.
First time 4.9.0 with internet no problems. Unchecked send print information  and check for updates in Preferences.
Second time 4.9.0. All problems are back! (4.9.0_WithInternet.log)
Third time no internet all OK. (4.9_noInternet.log)

[4.9.0_WithInternet.zip](https://github.com/Ultimaker/Cura/files/6727537/4.9.0_WithInternet.zip)
[4.9_noInternet.zip](https://github.com/Ultimaker/Cura/files/6727557/4.9_noInternet.zip)

GrooverFromHolland, 2021-07-03:
I solved the issue! It had nothing to do with Cura. I think Cura 4.9 and higher are more CPU speed sensitive. 
My “lovely” Dell BIOS decided to throttle down the CPU speed to 0.84 GHz, because of falsely detecting my original Dell power adapter as non-Dell!
Sorry for the time spend on this.

closed 2021-07-03 by GrooverFromHolland