Ultimaker/Cura #9707: Crashes every time I open a file with the Folder icon

dryphi, 2021-04-27:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Creality CR-6 SE

### Reproduction steps

First install:
1. Installed the newest version of the program (4.8 was working fine).
2. Clicked "yes" to uninstall the previous version at the beginning of the install.
3. Install proceeded as expected.
4. On first run, clicked "yes" to delete previous profiles.
5. When I clicked the folder icon to open a file, the program crashed.

Second install:
1. Uninstalled previous instance
2. Deleted everything in %APPDATA%\cura
3. Deleted anything related to Cura in Program Files
4. Ran CCleaner to remove registry entries
5. Installed v4.9 again

### Actual results

Program crashes every time I click the folder icon and navigate to a file

### Expected results

Program should open the file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6382214/cura.log)

No additional information to report.

dryphi, 2021-04-27:
This seems to be related to use of qt5core.dll as reported here:
https://github.com/smartavionics/Cura/issues/66#issuecomment-655699145
I uninstalled ChituBox and the problem has apparently disappeared.

dryphi, 2021-04-27:
There's a comment now on the ChituBox forum:
https://forum.chitubox.com/?thread-419.htm

Uninstall steps
1. Open cmd with administrator rights
2. Enter the command: regsvr32 / u "path (CHITUBOX_Thumbnail.dll path)"
3. Restart the resource manager

fvrmr, 2021-04-29:
Hi @dryphi thank you for your bug report.
This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/9648
So I will close this one since it is a duplicate.

But thank you for linking the forum of ChituBox!

closed 2021-04-29 by fvrmr