Ultimaker/Cura #9708: Cura 4.8 and 4.9 freeze upon startup

silas2471, 2021-04-27:
### Application Version

4.8 and 4.9

### Platform

Windows 10

### Printer

Artillery Genius

### Reproduction steps

Hey 
Yesterday I encountered an issue with Cura, where it freezes up at startup. After loading the UI i get presented a blank windows which doesn not respond to anything. This happened to me with Cura 4.8.
I tried:
- restarting the pc,
- reinstalling Cura (and upgrading to 4.9)
- removing connection to the internet
- removing unnecessary USB drives

I am using the default Cura without any extensions or such. 

### Actual results

A Blank UI Screen

### Expected results

The normal Cura interface.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Logs for both Cura 4.8 and 4.9 are attached.

[cura4.9.log](https://github.com/Ultimaker/Cura/files/6383320/cura4.9.log)
[cura4.8.log](https://github.com/Ultimaker/Cura/files/6383321/cura4.8.log)

fieldOfView, 2021-04-27:
Do you use multiple displays? If so, try disabling all but one of them temporarily. If Cura starts succesfully with a single monitor, go to the General preferences and uncheck the `Restore window position on start`. Then you can re-enable the other screens, and Cura should always open on your primary screen.

silas2471, 2021-04-27:
@fieldOfView Thanks for the quick help,
yes I am using two monitors and it fixed the problem. 
Interestingly it was no issue for the first few weeks of using it with the double monitor.
Possibly the Cura Devs should uncheck it as default?
Regards,
Silas

closed 2021-04-27 by silas2471

Ghostkeeper, 2021-05-13:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4427 then. I hoped that the Qt update would fix it, but alas.

> Possibly the Cura Devs should uncheck it as default?

That would cause the window state to be forgotten for everyone. It seems to be quite a rare problem caused by the unusual set-up of rendering each display with a different graphics card.

usernametakenaaaa, 2022-05-28:
The restore Window positions solution only helps to start Cura-5 but it will still open dialog windows on the wrong monitor.
It is not related to two graphics cards, I changed my setup to be on a single GPU (both monitors on one) and it's still happening.

Some of the QT dialogs are opening on the the wrong screen (the settings dialog works for example) Marketplace does not and some confirmations.
It's a combination of multiple bugs in Cura 5.