Ultimaker/Cura #9709: Application crash when choosing Scale

Petje1972, 2021-04-27:
### Application Version

4.9.0

### Platform

Linux - Zorin OS 64bit

### Printer

Custom (EasyThreed Nano)

### Reproduction steps

1. Downloaded project from https://www.thingiverse.com/thing:934033
2. Unzipped file
3. Opened Cura
4. Loaded stl file
5. Selected fox
6. Pressed scale
7. Application crashed

### Actual results

Application crashed / disappears

### Expected results

That the application does not crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional info

fieldOfView, 2021-04-27:
We really need logs to be able to diagnose this.

Petje1972, 2021-04-27:
Could you pleae inform me where I could find the logs?

fieldOfView, 2021-04-27:
See https://github.com/Ultimaker/Cura/#logging-issues

Petje1972, 2021-04-27:
Hi filedOfView,

I have found the log file, cleaned it and performed steps one time.
[cura.log](https://github.com/Ultimaker/Cura/files/6386303/cura.log)

Grtz Patrick

AccessDG, 2021-04-29:
I am encountering this as well

fieldOfView, 2021-05-01:
@AccessDG do you also use Zorin OS?

@Petje1972 do you use the AppImage, or another way to install Cura?

Petje1972, 2021-05-01:
@fieldOfView : I am using AppImage

smartavionics, 2021-05-02:
You could try my release which is built a little differently from the UM release. It can be found at https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420.AppImage. It won't affect your existing configuration files. If you do try it and have any problems, please report an issue at https://github.com/smartavionics/Cura/issues.

T-rex2017, 2021-05-02:
facing the same issue on Ubuntu 21.04 ,installed through latest app image from um website 
[cura.log](https://del.dog/mestehocru.txt) here is a possible clean log 
loaded an stl clicked on scale button and crash
does the app image need any dependencies 
and here is the part from loading the stl from the log file 
`[MainThread] cura.CuraApplication.readLocalFile [1759]: Attempting to read file file:///home/tony/Downloads/bfp_touch_remix.stl`

`[JobQueWorker [1]] STLReader.STLReader.load_file [61]: Using legacy code to load STL data.`

`[JobQueWorker [1]] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0003421306610107422 seconds`

`[JobQueWorker [1]] STLReader.STLReader._read [105]: Loaded a mesh with 2964 vertices`

`[JobQueWorker [1]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0020258426666259766 seconds. 62 input vertices. 62 output vertices.`

`[JobQueWorker [1]] UM.FileHandler.ReadFileJob.run [75]: Loading file took 0.0 seconds`

`[MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///tmp/.mount_UltimasI4won/usr/bin/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null`

`[JobQueWorker [1]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.007663249969482422 seconds. 62 input vertices. 62 output vertices.`

`[MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x9ea0650>, took 1ms`

`[MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'bfp_touch_remix.stl'>,rot.=Q<0.0,0.0,0.0,w=4.0>) TranslateOp.(node=<CuraSceneNode object: 'bfp_touch_remix.stl'>,trans.=<0.0,0.0,0.0>), took 8ms`

`[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_UltimasI4won/usr/bin/resources/shaders/transparent_object.shader]...`

`[MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles`

`[MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/tmp/.mount_UltimasI4won/usr/bin/resources/shaders/toolhandle.shader]...`

smartavionics, 2021-05-02:
Could you please also post the contents of the stderr.log file that should be in the directory above cura.log. Thanks...

T-rex2017, 2021-05-02:
> Could you please also post the contents of the stderr.log file that should be in the directory above cura.log. Thanks...

[stderr.log](https://del.dog/thocallame.txt)
here is the file

T-rex2017, 2021-05-02:
> You could try my release which is built a little differently from the UM release. It can be found at [...]

tried this build , has the same behaviour 
here is it's log file 
[cura.log](https://del.dog/ymimuppyly.txt)

T-rex2017, 2021-05-02:
> > Could you please also post the contents of the stderr.log file that should be in the directory above cura.log. Thanks...
> 
> [stderr.log](https://del.dog/thocallame.txt)
> here is the file

i don't know if this is important or not 
i had to mess with some python files in order to make pronterface working , mainly i had to build wxPython for Ubuntu21.04
it was mostly in a virtual environment (venv), anyways i will try this on a live boot of the same distro and see how it behaves

smartavionics, 2021-05-02:
The stderr.log actually contains some useful info re the crash...
```
File "/opt/cura/lib/python3.8/site-packages/scipy-1.6.1-py3.8-linux-x86_64.egg/scipy/linalg/lapack.py", line 1005 in _compute_lwork
  File "/opt/cura/lib/python3.8/site-packages/scipy-1.6.1-py3.8-linux-x86_64.egg/scipy/linalg/decomp_svd.py", line 121 in svd
  File "/tmp/.mount_Cura-mSq8xJB/usr/bin/plugins/Tools/ScaleTool/ScaleTool.py", line 470 in _getSVDRotationFromMatrix
  File "/tmp/.mount_Cura-mSq8xJB/usr/bin/plugins/Tools/ScaleTool/ScaleTool.py", line 499 in _getRotatedExtents
  File "/tmp/.mount_Cura-mSq8xJB/usr/bin/plugins/Tools/ScaleTool/ScaleTool.py", line 458 in _getScaleInWorldCoordinates
  File "/tmp/.mount_Cura-mSq8xJB/usr/bin/plugins/Tools/ScaleTool/ScaleTool.py", line 298 in getScaleX
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Qt/Bindings/ActiveToolProxy.py", line 117 in _updateProperties
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Qt/Bindings/ActiveToolProxy.py", line 107 in _onActiveToolChanged
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Signal.py", line 332 in __performEmit
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Signal.py", line 219 in emit
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Controller.py", line 351 in setActiveTool
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 53 in setActiveTool
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 187 in mouseReleaseEvent
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 383 in exec_
  File "/home/markb/3dp/cura/cura-build.git/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 862 in run
  File "/home/markb/3dp/cura/cura-build.git/build/inst/bin/cura", line 269 in <module>
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36 in run
  File "/opt/cura/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74 in run
```

It's not my area of expertise but perhaps this will be useful to the Cura devs.

I'm running that same AppImage on the system it was built on with no issues detected.

T-rex2017, 2021-05-02:
> i don't know if this is important or not
> i had to mess with some python files in order to make pronterface working , mainly i had to build wxPython for Ubuntu21.04
> it was mostly in a virtual [...]

tested on a live boot of Ubuntu 21.04 and the results are same

Petje1972, 2021-05-02:
@smartavionics : I have tested the Cura-mb-master-x86_64-20210420.AppImage but this version alo crashes.
Here is my [cura.log](https://github.com/Ultimaker/Cura/files/6411095/cura.log) file and my [stderr.log](https://github.com/Ultimaker/Cura/files/6411103/stderr.log) file.

I also tested it with an old version I downloaded some while ago (cura version 4.7.1), this one does not crash when clicking the scale button.
For comparence I added [cura.log](https://github.com/Ultimaker/Cura/files/6411110/cura.log) and [stderr.log](https://github.com/Ultimaker/Cura/files/6411112/stderr.log) for version 4.7.1

T-rex2017, 2021-05-02:
> tested on a live boot of Ubuntu 21.04 and the results are same

I'm currently trying to build it from source and see how that goes

smartavionics, 2021-05-02:
I have built a release that uses the version of scipy used for the earlier Cura releases, please give it a go and see if the crash still occurs...

https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-1.AppImage

T-rex2017, 2021-05-02:
> I have built a release that uses the version of scipy used for the earlier Cura releases, please give it a go and see if the crash still occurs...
> 
> https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-1.AppImage
> 

Same result , crashes , do you want see the logs ?

smartavionics, 2021-05-02:
Thanks for trying it out. Don't bother with the logs, it surely will be the same.

smartavionics, 2021-05-02:
One more try. This build uses an earlier version of numpy. It's not the same as used for Cura 4.8 but it may make a difference. I'm not confident that it will be any different.

https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-2.AppImage

T-rex2017, 2021-05-02:
> One more try. This build uses an earlier version of numpy. It's not the same as used for Cura 4.8 but it may make a difference. I'm not confident that it will be any different.
> 
> https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-2.AppImage

i could try , but I'm printing something , cura usb print plugin will interrupt the print , is there a way to disable it without opening the app

Petje1972, 2021-05-02:
@smartavionics, I have tried the 20210420-2 version, this version also cashes on pressing scale button.

smartavionics, 2021-05-02:
OK, let's see if we can get more info...

Run one of the bad AppImages but don't let it crash. In a terminal enter "ps aux | grep cura" to find the PID of the running cura process and then enter "lsof -p PID" where PID is the number you just discovered. Please post the resulting output here. Thanks.

T-rex2017, 2021-05-02:
i assume this is what you need
[lsof.log](https://del.dog/opylubiram.txt)

> tony      241895  0.0  0.0   2632   716 ?        S    19:26   0:00 /bin/sh /tmp/.mount_UltimaEIBrIN/usr/bin//cura.sh
tony      241911  2.0  3.3 2174636 265968 ?      Sl   19:26   0:07 cura
tony      263653  0.0  0.0  17768   844 pts/0    S+   19:32   0:00 grep --color=auto cura

these where the op of `ps aux | grep cura` , i used the 2nd one

smartavionics, 2021-05-02:
Yes, thanks, that's just what I wanted. Unfortunately, there's nothing there that looks like it could cause trouble.

T-rex2017, 2021-05-02:
Do you want it to be crashed after getting pid , then get the lsof , i didn't let it crash before taking the log , ( isn't the pid be gone ones the app crashes ?¿)

smartavionics, 2021-05-02:
Yes, the process (and its pid) will be gone once it crashes so what you did was fine. I wanted to see the libraries it was using just in case it was using some incompatible library on your system. As far as I can tell, it looked OK.

Petje1972, 2021-05-02:
@smartavionics , here is mine 
[lsof.log](https://github.com/Ultimaker/Cura/files/6411605/lsof.log)

smartavionics, 2021-05-02:
Thanks @Petje1972, nothing obviously bad in there.

smartavionics, 2021-05-02:
Hi, I have made another release that uses an earlier version of another library (openblas), I don't expect it will make any difference but could someone please try this, thanks.

https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-3.AppImage

T-rex2017, 2021-05-02:
> Hi, I have made another release that uses an earlier version of another library (openblas), I don't expect it will make any difference but could someone please try this, thanks.
> 
> https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-3.AppImage

WORKING 
finally something that work , congrats & thanks for solving the issue

smartavionics, 2021-05-02:
That's interesting, what computer are you using? I mean what CPU chip?

T-rex2017, 2021-05-02:
> That's interesting, what computer are you using? I mean what CPU chip?

it's Intel Pentium G3250

Petje1972, 2021-05-02:
@smartavionics : No crashes for me with this version (-:

Intel(R) Pentium(R) Gold G5420 CPU @ 3.80GHz

T-rex2017, 2021-05-02:
do you know the cause of the issue , what is the difference between the lib that you used on this build and the one percent in 4.90

T-rex2017, 2021-05-02:
> @smartavionics : No crashes for me with this version (-:
> 
> Intel(R) Pentium(R) Gold G5420 CPU @ 3.80GHz

that's Intrusting we both are using Intel Pentium chips , does that last test build worked fine for you ?

Petje1972, 2021-05-02:
> > @smartavionics : No crashes for me with this version (-:
> > Intel(R) Pentium(R) Gold G5420 CPU @ 3.80GHz
> 
> that's Intrusting we both are using Intel Pentium chips , does that last test build worked fine for you ?

I just clicked through the application without any problems.
Dit not test it creating a gcode file

smartavionics, 2021-05-02:
My guess is that the more recent version of the library is enabling by default a CPU instruction that is not supported on those chips you guys are using. I'm busy now till tuesday so I'll look into it then. Still, the good news is that my build works now for you and you can have fun discovering all the extra goodies that it contains compared to the UM release!

Petje1972, 2021-05-02:
Thanks @smartavionics for the solution.

treii28, 2021-05-02:
Getting the same issue on all three of my linux systems. Tried uninstalling all plugins. The machines in question are a mix of an i7 and two amd processors.

smartavionics, 2021-05-02:
> Getting the same issue on all three of my linux systems. Tried uninstalling all plugins. The machines in question are a mix of an i7 and two amd processors.

Does my -3 build work?

fvrmr, 2021-05-04:
I have discussed this issue with the team. And we decided to defer this issue, since we can't reproduce it ourselves.

smartavionics, 2021-05-04:
> I have discussed this issue with the team. And we decided to defer this issue, since we can't reproduce it ourselves.

So what you're saying is that you no longer support Cura running on Pentium processors. Don't forget to advertise that in your hardware requirements for Cura.

fieldOfView, 2021-05-04:
But @smartavionics gave you the solution (packaging an older version of openblas)!

smartavionics, 2021-05-04:
BTW, Here's how you can fix this issue...
```
if(NOT BUILD_OS_WINDOWS)
    # Fortran compiler is needed for OpenBLAS, but it does no check whether it is available.
    enable_language(Fortran)
    if(${CMAKE_CXX_LIBRARY_ARCHITECTURE} MATCHES "arm-linux-gnueabihf")
        set(openblas_options DYNAMIC_ARCH=1 NO_STATIC=1)
        set(openblas_build_targets libs netlib shared) # don't do tests as something is causing kernel crashes
    else()
        set(openblas_options DYNAMIC_ARCH=1 NO_STATIC=1 TARGET=HASWELL)
        set(openblas_build_targets all)
    endif()

    if(${CMAKE_CXX_LIBRARY_ARCHITECTURE} MATCHES "x86_64-linux-gnu")
        # newer openblas uses illegal instruction on Pentium CPUs
        set(openblas_url https://github.com/xianyi/OpenBLAS/archive/v0.3.7.tar.gz)
        set(openblas_md5 5cd4ff3891b66a59e47af2d14cde4056)
    else()
        set(openblas_url https://github.com/xianyi/OpenBLAS/archive/v0.3.13.tar.gz)
        set(openblas_md5 2ca05b9cee97f0d1a8ab15bd6ea2b747)
    endif()

    ExternalProject_Add(OpenBLAS
        URL ${openblas_url}
        URL_MD5 ${openblas_md5}
        CONFIGURE_COMMAND ""
        BUILD_COMMAND make PREFIX=${CMAKE_INSTALL_PREFIX} ${openblas_options} ${openblas_build_targets}
        INSTALL_COMMAND make PREFIX=${CMAKE_INSTALL_PREFIX} ${openblas_options} install
        BUILD_IN_SOURCE 1
    )
else()
    add_custom_target(OpenBLAS)
endif()

SetProjectDependencies(TARGET OpenBLAS)
```

There, i've done your work for you...

rakoenig, 2021-05-04:
> > I have discussed this issue with the team. And we decided to defer this issue, since we can't reproduce it ourselves.
> 
> So what you're saying is that you no longer support Cura running on Pentium processors. Don't forget to advertise that in your hardware requirements for Cura.

Not only "Pentium". I have this issue here on a dual CPU Xeon workstation:

$ lscpu
Architektur:                   x86_64
CPU Operationsmodus:           32-bit, 64-bit
Byte-Reihenfolge:              Little Endian
Address sizes:                 40 bits physical, 48 bits virtual
CPU(s):                        24
Liste der Online-CPU(s):       0-23
Thread(s) pro Kern:            2
Kern(e) pro Socket:            6
Sockel:                        2
NUMA-Knoten:                   2
Anbieterkennung:               GenuineIntel
Prozessorfamilie:              6
Modell:                        44
Modellname:                    Intel(R) Xeon(R) CPU           X5670  @ 2.93GHz
Stepping:                      2
CPU MHz:                       2642.174
Maximale Taktfrequenz der CPU: 2927,0000
Minimale Taktfrequenz der CPU: 1596,0000
BogoMIPS:                      5851.45
Virtualisierung:               VT-x
L1d Cache:                     32K
L1i Cache:                     32K
L2 Cache:                      256K
L3 Cache:                      12288K
NUMA-Knoten0 CPU(s):           0-5,12-17
NUMA-Knoten1 CPU(s):           6-11,18-23
Markierungen:                  fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid dtherm ida arat flush_l1d

smartavionics, 2021-05-04:
Good point @rakoenig, did you try my -3 build mentioned above?

rakoenig, 2021-05-04:
> Good point @rakoenig, did you try my -3 build mentioned above?

A few days ago I downloaded Cura-mb-master-x86_64-20210420-3.AppImage and this works for me.

smartavionics, 2021-05-04:
So the solution is clear, they should revert to using the earlier OpenBLAS version if they wish for future Cura releases to be usable on a wide range of processors running Linux.

T-rex2017, 2021-05-05:
> So the solution is clear, they should revert to using the earlier OpenBLAS version if they wish for future Cura releases to be usable on a wide range of processors running Linux.

Could the people from openBLAS help to fix the problem with newer versions

Ghostkeeper, 2021-05-05:
The workaround provided by Smartavionics here is not something we'd like to adopt because we prefer to keep the versions of our dependencies the same across all platforms. But it does help a lot to know that apparently downgrading to 0.3.7 works around the problem. We'll adopt something similar for release 4.10.

On my computer I have the architecture `x86_64-linux-gnu` though and I'm not getting the crash with the OpenBLAS I'm compiling (nor with the version used in the build), so apparently the conditions for needing to downgrade are not entirely correct.

rburema, 2021-05-05:
In addition to what @Ghostkeeper said, here's the internal ticket nr. of the ticket that was just made for it so us (internal) devs can trace it back when needed: CURA-8215

T-rex2017, 2021-05-06:
Btw i'm on wayland , 21.04 of Ubuntu uses it by default , does this cause any issues

smartavionics, 2021-05-06:
I guess it may but I don't think this particular issue will be affected by using Wayland.

jk101290, 2021-05-14:
Hi guys I have the same problem on Linux Mint 20

[cura.log](https://drive.google.com/file/d/1oN3feChVm7MHv8kw3gapibFMCw3VAcuY/view?usp=sharing)

stevedee, 2021-05-15:
I have same problem on Linux PeppermintOS running Cura 4.9.0/4.9.1:-
System:    Kernel: 4.15.0-143-generic x86_64bits: 64 gcc: 7.5.0  Desktop: N/A Distro: Peppermint Nine
CPU:       Quad core Intel Celeron N3450 (-MCP-) arch: N/A cache: 1024 KB flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 8755  clock speeds: max: 2200 MHz 1: 1388 MHz 2: 1596 MHz 3: 1495 MHz 4: 1532 MHz

stderr.log extract:-
Fatal Python error: Illegal instruction

Thread 0x00007f7154ff9700 (most recent call first):
  File "/tmp/.mount_4.9_20YgFWmr/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f7155ffb700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 306 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 1434 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f71567fc700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 306 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 558 in wait
  File "/tmp/.mount_4.9_20YgFWmr/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f7156ffd700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 306 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 1262 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f71577fe700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/site-packages/zeroconf/__init__.py", line 1183 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f7157fff700 (most recent call first):
  File "/tmp/.mount_4.9_20YgFWmr/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f7174ff9700 (most recent call first):
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f71757fa700 (most recent call first):
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 870 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f71967fc700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 302 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 124 in run
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f7196ffd700 (most recent call first):
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 302 in wait
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
[... 14 lines cut ...]
  File "/srv/cura-build-environment/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
[... cut ...]
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 383 in exec_
  File "/home/ultimaker/src/output/build/inst/lib/python3.8/site-packages/cura/CuraApplication.py", line 862 in run
  File "/home/ultimaker/src/output/build/inst/bin/cura", line 241 in <module>
  File "/srv/cura-build-environment/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36 in run
  File "/srv/cura-build-environment/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74 in run

jsn0327, 2021-05-17:
> Hi, I have made another release that uses an earlier version of another library (openblas), I don't expect it will make any difference but could someone please try this, thanks.
> 
> https://github.com/smartavionics/Cura/releases/download/20210420/Cura-mb-master-x86_64-20210420-3.AppImage

I had the same issue where clicking scale would kill Cura 4.9.1.  Your fix worked for me too, thanks!
Ubuntu 20.04.2 LTS
AMD Phenom(tm) II X4 965 Processor

X-Ryl669, 2021-05-18:
I've the same issue too with a Core i7 CPU. If it's an OpenBLAS issue, I think we should report to OpenBLAS so it's fixed in a later release. I don't think downgrading is a long term solution.

X-Ryl669, 2021-05-18:
It's linked with [this OpenBLAS issue](https://github.com/xianyi/OpenBLAS/issues/3139). TL;DR; Starting with OpenBlas 0.3.13, it does not follow the configuration flag DYNAMIC_ARCH and instead uses the host's CPU for deducing compiler flags. Obviously @fvrmr, it works on your build system since it's using the CPU features of **your** system. 

The possible solution for this bug would be to build OpenBLAS with [the right flags](https://github.com/flathub/org.octave.Octave/pull/132/files), that is DYNAMIC_ARCH, DYNAMIC_OLDER, TARGET=GENERIC as stated in this [Octave's bug report](https://github.com/flathub/org.octave.Octave/issues/131)

[22 more replies]

closed 2021-05-25 by nallath