Ultimaker/Cura #9710: Cura crashes after running untouched for long time

mbourque, 2021-04-27:
### Application Version

4.9.0

### Platform

OSX

### Printer

Creality CR-6 SE

### Reproduction steps

1. Open Cura
2. Load model
3. Slice
4. Leave it alone

### Actual results

Crashes

### Expected results

No crash

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

2021-04-25 23:51:44,648 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:00:54,659 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:00:55,123 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:01:05,456 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:10:00,367 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:10:02,276 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:10:12,453 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:10:13,449 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x6000073b7df0>] timeout.
2021-04-26 00:10:13,453 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[741f15db][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-26 00:10:13,459 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[741f15db][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-26 00:10:13,460 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
2021-04-26 00:19:24,649 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:19:25,123 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:19:40,348 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:28:40,451 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:28:40,966 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:28:54,641 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:37:54,557 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:37:55,331 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:38:10,460 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 00:47:03,458 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 00:47:03,991 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 00:47:14,450 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 04:50:46,101 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 04:50:46,835 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 04:51:00,336 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 08:51:29,603 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 08:51:30,748 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 08:51:45,161 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 09:00:58,290 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-26 09:00:58,847 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-26 09:01:18,245 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-26 09:08:28,063 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x6000073b7df0>] timeout.
2021-04-26 09:08:28,069 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[09d032a9][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-26 09:08:28,074 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[09d032a9][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-26 09:08:28,075 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
[... 361 lines cut ...]
2021-04-26 16:13:33,333 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[223dc30d][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/connect/v1/clusters?status=active')][timeout=10][no-data] was aborted, do nothing
2021-04-26 16:13:33,343 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[ab53c191][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-26 16:13:33,349 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[ab53c191][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-26 16:13:33,350 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
2021-04-26 16:14:14,709 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x6000073b7df0>] timeout.
2021-04-26 16:14:14,715 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[1602636d][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/connect/v1/clusters?status=active')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The  [... cut ...]
2021-04-26 16:17:48,348 - DEBUG - [MainThread] UM.Logger.debug [122]: User is not logged in for Cloud API request to https://api.ultimaker.com/connect/v1/clusters?status=active
2021-04-26 16:17:48,355 - WARNING - [MainThread] cura.OAuth2.AuthorizationService.refreshAccessToken [141]: Unable to refresh access token, since there is no refresh token.
2021-04-26 16:17:48,361 - DEBUG - [MainThread] UM.Logger.debug [122]: User is not logged in for Cloud API request to https://api.ultimaker.com/cura-packages/v1/user/packages

fvrmr, 2021-04-30:
Hi @mbourque thank you for your bug report.
I can't reproduce this issue, what computer do you have?

mbourque, 2021-05-03:
OSX Big Sur

mbourque, 2021-05-14:
It now crashes all the time on 4.9.1. Sure I understand, but I think it is a bad situation as I need to rollback to 4.8.

Ghostkeeper, 2021-05-18:
Quite a few people seem to be experiencing this so far. I think it would be good to fix this, but we've tried to reproduce this to no avail. The error says "out of resources" so that may be RAM or a thread limit, but we're not seeing that. Those things can be very computer-dependent. Until we get more of a hint on what sort of situations it crashes in, there's not a lot we can do. But if more people are able to share their experiences there may arise a pattern from which we can deduce how to fix this.

jellespijker, 2021-05-19:
@Ghostkeeper I agree that this is something to be fixed. It is unclear to me what the original motivation to defer it was. 

_A trimmed down comité for the eCCB has its downside_

Ghostkeeper, 2021-05-31:
It was deferred because we have no basis on how to begin debugging this yet. It would be "needs more information" except that the bot would close the issue in 2 weeks then.

mbourque, 2023-05-26:
It still crashes on latest version for macOS Regards,Michael Bourque

MariMakes, 2023-06-05:
Hey @mbourque,

A fresh set of eyes reporting for duty 👋 

We still get reports on this, I'll mark this to be a duplicate of https://github.com/Ultimaker/Cura/issues/15161
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15161

I hope we can resolve this soon. 
Thanks again 👍

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

mbourque, 2024-11-19:
Fixed Regards,Michael Bourque

closed 2024-11-19 by GregValiant