Ultimaker/Cura #9712: Extruder 1 material delayed in Cura 4.9

kevinjones4642, 2021-04-27:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I have parts that I sliced sliced with Cura 4.9 using both ABS and PLA, both with the 0.25mm extruder in slot 1, where the material in extruder 1 seems to retract pretty far, and doesn't start to actually extrude material for the first several minutes of a job. Extruder 2, with a 0.4mm nozzle and breakaway material works fine. I've tried unloading/reloading the material just prior to starting a job, but get the same results. Out of desperation, I've waited for the 1st layer of breakaway material to finish printing, and then quickly manually released the feed tensioner, forced material in, and lowered the feed lever, and then the print is fine. I though it might be a firmware issue. That printer was still back on something like 5.4, so I updated to 6.3, but got the same results. I just rolled back to Cura 4.8, sliced the same part, and the part prints perfectly, so I am assuming this is an issue with Cura 4.9.

### Actual results

Material in extruder 1 seems to retract at start of print job, such that 1st layer(s) are missing, and part fails. Extruder 2 seems fine.

### Expected results

Part should print correctly. I've attached the part sliced in Cura 4.9. I can send the same part sliced in Cura 4.8 if needed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_PixRacer Softmount AQ V1.2 Cura 4.9.3mf.zip](https://github.com/Ultimaker/Cura/files/6386169/UMS5_PixRacer.Softmount.AQ.V1.2.Cura.4.9.3mf.zip)
 Project file for part sliced in Cura 4.9.

nallath, 2021-04-28:
We found that regression ourself as well. A fix is already in review: https://github.com/Ultimaker/CuraEngine/pull/1445

closed 2021-04-28 by nallath

kevinjones4642, 2021-05-07:
Hi Jaime,

Is there a link to an install file with this fix?

--Kevin

nallath, 2021-05-10:
No, not yet.

Ghostkeeper, 2021-05-13:
The fix has been released with Cura 4.9.1, so here is a link: https://github.com/Ultimaker/Cura/releases/tag/4.9.1