Ultimaker/Cura #9714: [4.9.0]

ab-tools, 2021-04-27:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Anycubic Mega S

### Reproduction steps

1. Load attached "fan_case_outsidecover.stl".
2. Slice.

### Actual results

The large grid in the middle of the object is sliced in a way that it does not get printed at the bottom layer (but at a higher layer "in the air" although it is laying on the ground in the STL file.

### Expected results

The grid is printed starting from the very first layer as it is connected to the ground.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Source STL and project file attached:
[AI3M_fan_case_outsidecover.zip](https://github.com/Ultimaker/Cura/files/6387612/AI3M_fan_case_outsidecover.zip)

Grid clearly connected to the ground in the source STL file:
[image]

But not printed in first/bottom layer according to slicer:
[image]

GregValiant, 2021-04-27:
The lines that make up the grid are too fine for your Initial Layer Horizontal Expansion of -0.3 and the Initial Layer Line Width of 160% of .38 (0.6).  Those settings just aren't compatible with the geometry of the part.
With Initial Layer Horizontal Expansion at 0 and Initial Layer Line Width at .38, it's fine.

Ghostkeeper, 2021-04-30:
The pattern here is 0.866mm wide, according to my measurement in Blender:

[image]

Your line with is 0.35, but on the first layer it's set to 160%, so that would become 0.608mm. The Initial Layer Horizontal Expansion is set to -0.3mm, so that takes away 0.6mm in total (0.3 from either side), leaving you with a part of 0.266mm. Since the line width is 0.608, the line doesn't fit in the part and you get no pattern in the first layer.

fvrmr, 2021-04-30:
So that's why this is expected behaviour why the first layer isn't printed. Let me know if this makes sense to you.

ab-tools, 2021-05-01:
Thanks for the quick feedback everybody!

That's understood, thanks for the explanations.

Would it perhaps make sense to somehow inform the user about this problem during slicing?
Skipping the first layer obviously just creates a mess when trying to print that, so even considering the slicing process as "failed" when the current slicing settings do not allow to print the first layer of a major part of the model might make sense.

At least showing some sort of warning message to the user would be helpful and prevent an unseuccssfull and frustrating "print try". ;-)

Thanks again for the support
Andreas

GregValiant, 2021-05-01:
It's model specific and one of those things that even an AI would have trouble with.  The question would be "Is this what the user wanted?"
So you get the visual clue instead and get to ask that age old question "What the #$%@ is going on around here?"

ab-tools, 2021-05-01:
True, I understand what you mean, but the visual cue that you mentioned is only visible when you zoom in a lot on the preview to detect that the first layer is missing. I didn't do that before I printed it, because I had no sign that something could potentially be wrong.

So first I printed the model, noticed that it printed "in the air" and only then I searched in the model for the reason.

Therefore, to prevent this unneeded "print try", an information somewhere that the first layer cannot be fully printed with the current settings would have helped me a lot.

Ghostkeeper, 2021-05-13:
If we were to give visual cues on overhangs, you'd see those cues all over the place and you'd quickly learn to ignore them.

Cura would of course generate support for overhang like this. Only the support needs to keep a certain Z distance, to ensure that it doesn't fuse with the model. This necessarily allows the model to sag by a few tenths of a millimetre. So to prevent this issue, you could reduce the Support Z Distance setting at the cost of fusing support to the print more strongly. Otherwise, Cura thinks that sagging by this one layer is an acceptable consequence. Yes, it may sag, but that's apparently not a problem.

Of course, in reality it is a problem, but it all boils down to optimising settings then. Cura thinks this is what you wanted.

I'll close this as expected behaviour.

closed 2021-05-13 by Ghostkeeper