Ultimaker/Cura #9718: pulling old code out of thin air

WOWElectronics, 2021-04-28:
### Application Version

appimage for amd64, version 4.9.0

### Platform

linux debian buster

### Printer

Ender 5 Plus

### Reproduction steps

1.sent cura 4.9.0 a 93686k .stl from openscad.. trying to make a ball bearing assembly for crosman bb's as the bearings balls for a bearing 98mm in diameter. because I could see and feel the straight line facets, and the balls were obviously too big for the groove, i added .01mm to the size of the balls groove 4 times and increased the value of openscad's $fn from 360= 1 degree wide, to 720 and 1440 by try 3, making the facet size 1/4 of a degree, but bed adhesion failed on pass 4.

It was on pass 4 that I noted the resliced and resaved produced gcode file had not grown, nor had its creation date/time changed and despite backing out of the directory and re-reading it, the file produced did not grow with my changes, and the rendered plastic remained a way too tight missfit. mc had no problem looking at the same file. mc has no  problem seeing the saved, bigger gcode file, but stuffed to the u-sd card and sneaker netted to  the printer, the CE5P_ prefixed to the gcode filename also renders the file invisible to the ender's print menu.

How should I proceed?

### Actual results

the produced plastic does not contain my adjustments.

### Expected results

I expected the rendered groove for the balls to grow until it fit.  And I had done this successfully with the 4.8.0 AppImage

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The openscad .scad file:

// my feeble attempt at generating a ball bearing using bb's.
// my alu block is 140.9mm square, so this will have
// to shrink as this bearing is now 148.3mm, or find wider
// stock
dmtr=98.00; // in mm's of ID,squeeze it a hair.
height=6.33; // in mm's
// everything below is generated center=true
// which means we're dealing in radius
ctr_ht=(height * .50000); // mults are faster
halfd=(dmtr *.50000); // rad of inner diameter
echo(11, halfd);
innr=(halfd+ctr_ht); // sub to move innr ID to dmtr
echo(13, innr);
outd=(dmtr+height*2.000);// outer face dia 147.6mm
echo(15, outd);
bb=4.3725; // .005mm more than a crosman bb so its not
// preloaded but snug. Smaller= tighter fit
// 4.3625 way tite, next=4.3725 making now
// how many bb's?
racedia=dmtr+ctr_ht;
racircum=racedia*PI;
echo(22, bb, racedia, racircum);
bbs=racircum/bb;
echo(24, bbs);
$fn=1440;
//echo(25, bb, $fn);

rotate_extrude()
    translate([innr,0,0]) 
    difference()
    {
        square(height, center = true);
        circle(d=bb);
        square([.26*height, 1.1*height],center=true);
    }

GregValiant, 2021-04-29:
If you were running Windows I would say to start by shortening the file name to an 8character.GCODE format and try again.  The earlier Enders and CR-10's didn't understand file name prefixes longer than 16 characters.  They also didn't like SD cards over 32gb.

WOWElectronics, 2021-04-29:
I got that figured out when cura added a CE5P_ to the filename and brand new ender 5 plus couldn't see it. Thats just one of the reasons I'd like to see FAT filesystems fade into the sunset.

GregValiant, 2021-04-29:
There is an option under "Preferences/Configure Cura/Open and Saving" regarding adding the machine as a prefix to the file name.

WOWElectronics, 2021-04-29:
Thank you, a bunch, I was not aware off that. That will restore sneakernet file transportation.

fvrmr, 2021-04-30:
Hi @WOWElectronics thank you for your report.
I think why your file isn't getting bigger is probably because you've reached the `maximum resolution` and `maximum deviation`. 
About not saving did you try to save it on a different drive? And see if the file gets updated?

WOWElectronics, 2021-04-30:
On Friday 30 April 2021 07:51:59 Fenne wrote:

> Hi @WOWElectronics thank you for your report.
> I think why your file isn't getting bigger is probably because you've
> reached the `maximum resolution` and `maximum deviation`. About not
> saving did you try to save it on a different drive? And see if the
> file gets updated?

Whats wierd is that when saveing over the local net here st the 
coyote.den from this machine to the one driving the printer, is that 
when I then go to that machine to load the file into cura and slice it, 
the file size shows as 0 bytes.  Just for S&G, i tried to load that 
empty file, and it worked, its a 4mm thick test elipse disk being 
printed to verify its rim size is correct to be the pusher in a new 
design of a harmonic drive. If it is, those measurements get copied to 
another file that already has a hub defined, but it will be 19mm thick 
and much of a day to print.

Now, to make this report even more interesting, I backed out of that 
directory by two levels when I saw he zero length file, then came back 
to it hoping to flush the caches, but when I came back to that directoy 
it was still shown as zero length. It was at that point I clicked on it, 
loaded and sliced it, so it will be abut 2.5 hours printing a zero 
length .stl. This is the amd64 AppImage of cura-4.9.0. This machine is 
running an old but uptodate debian stretch and has 32GB of dram, while 
the machine driving the printer is an updodate buster with 16 gigs of 
dram, both i5's. Go figure, I'm fresh out of theories.

Cheers, Gene Heskett
-- 
"There are four boxes to be used in defense of liberty:
 soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author)
If we desire respect for the law, we must first make the law respectable.
 - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/gene>

Ghostkeeper, 2021-05-13:
Your "reproduction steps" are a confusing story of things you happened to think about when you were encountering a problem.

How would *I* be able to reproduce this problem? Could you make a numbered list? Like:

1. Compile this .scad file.
2. Export that to .stl.
3. Add a new Ender 5 Plus printer to Cura.
4. Load that STL into Cura.
5. Slice it and save the .gcode to a removable drive.
6. Check the file size of this file.

Those are the steps I tried, but it gives me a .gcode file of 4.8MB, not 0 bytes. And the file modification timestamp updates fine.

Give me a numbered list of actual steps. Really. I'll execute them like a robot and see if I can reproduce your problem. If not, we'll have to close this issue due to lack of information.

One thing that I'll suggest though is to try a different SD card. They wear out, especially cheap ones or big ones.

WOWElectronics, 2021-05-14:
On Thursday 13 May 2021 17:11:58 Ghostkeeper wrote:

> Your "reproduction steps" are a confusing story of things you happened
> to think about when you were encountering a problem.
>
> How would *I* be able to reproduce this problem? Could you make a [...]

Not normally using an sd card.

1; I generally do the design on this machine using openscad, which has 
32G of dram, a 6 core I5 and a comfy chair.

2; then using sshfs, copy the .stl to the printers driver computer (a 
dell 7010 with a 4 core I5 and 16GB of dram) in the other end of the 
house 

3; either by direct (sshfs) network access from openscad, or with mc, to 
the 500GB brand new SamSung SSD in that machine,

4; then load it into cura, which I just updated to 4.9.1, for slicing,

5; then "print to usb" plugged into the ender 5 plus on the next table.  
If that prints a usable print, I'll

6; then save the gcode. The file loads seemingly normally even if it says 
zaro bytes, and backing out of that directory, then re-entering it still 
doesn't update the erronious size.  The file is there, but something 
isn't updateing the cura cache properly. 

7; If I send mc out over the cable, to look at that same directory, all 
the hundred or so files there have valid sizes and dates. But cura 4.9.0 
isn't showing it in its file requestor. I haven't had 4.9.1 (linux 
AppImage) installed long enough to check that.

Cheers, Gene Heskett
-- 
"There are four boxes to be used in defense of liberty:
 soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author)
If we desire respect for the law, we must first make the law respectable.
 - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/gene>

WOWElectronics, 2021-05-14:
On Thursday 13 May 2021 17:11:58 Ghostkeeper wrote:

> Your "reproduction steps" are a confusing story of things you happened
> to think about when you were encountering a problem.
>
> How would *I* be able to reproduce this problem? Could you make a [...]

Cheers, Gene Heskett
-- 
"There are four boxes to be used in defense of liberty:
 soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author)
If we desire respect for the law, we must first make the law respectable.
 - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/gene>

Ghostkeeper, 2021-05-21:
I will start here with step 4 since the first 3 steps are not in Cura. So I must assume that the rest of your computer is in working order. So basically the reproduction steps become then (to summarise):

1. Add an Ender 5-Plus printer to Cura.
2. Load [this STL file](http://dulek.net/work/bearing.zip) into Cura 4.9.1.
3. Let it slice and print via USB cable.
4. Save the g-code to disk.
5. Verify the file size of the .gcode file created on disk.

I can only send it to an emulator in step 3, since I don't have your actual printer, so another assumption I must make is that the printer is behaving properly. For Cura whether or not it behaves properly it doesn't really matter though, and you would see that the print is incomplete if it didn't, so that's a pretty safe assumption.

The file I get after step 5 is 4.8MB. That seems correct to me.

So that means that one of my other assumptions must be wrong. Either the printer is not behaving correctly (unlikely) or the assumption at the start was incorrect that the first 3 steps are behaving correctly. So either your STL is not generating properly (also unlikely since your USB print was apparently successful) or your file system is not behaving correctly (more likely considering your cumbersome transfer steps).

What in the world could I do with this? Cura is not responsible for making sure that sshfs is working correctly on your computer. Am I making a wrong assumption somewhere? Am I summarising your reproduction steps incorrectly? What did I do wrong?

WOWElectronics, 2021-05-21:
On Friday 21 May 2021 05:47:15 Ghostkeeper wrote:

> I will start here with step 4 since the first 3 steps are not in Cura.
> So I must assume that the rest of your computer is in working order.
> So basically the reproduction steps become then (to [...]
Will not complete the load into cura 4.9.1 on this stretch machine,

Shows this as cura appimage is launching and loading untitled.stl:
***@***.***:~/AppImages$ cura
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated 
in favour of importlib; see the module's documentation for alternative 
uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is 
created.
Fontconfig warning: "/etc/fonts/fonts.conf", line 100: unknown 
element "blank"
Cyclic dependency detected 
between "file:///tmp/.mount_curaCMbLEv/usr/bin/resources/qml/Actions.qml" 
and "file:///tmp/.mount_curaCMbLEv/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - 
fallback: "/tmp/.mount_curaCMbLEv/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation

The second time I loaded in tnto cura it did succeed, and did slice it, 
saving the file to that machine. Except it didn't save to that machine, 
it saved it to the same place on this machine as a just sy of 10 
megabyte file.
But this machine is stretch, the target machine is buster.
So I copy that 300 some megabyte untitled.stl to the target machine.
And it shows as a 368some megabyte file there, from here.

So now I goto that machine, and launche cura, again from the AppImage,
and getting the same warnings as above, but it loads, runs and is now 
loading that copy. On a 16gigabyte, 4 core i5 machine that takes several 
minutes.

So I slice it, get 6620894 bytes of gcode.  Saved to disk, read by mc.

> 4.9.1. 3. Let it slice and print via USB cable.
> 4. Save the g-code to disk.
> 5. Verify the file size of the .gcode file created on disk.
>
> I can only send it to an emulator in step 3, since I don't [...]

Moderatly so, I may be set for higher resolution.

> So that means that one of my other assumptions must be wrong. Either
> the printer is not behaving correctly (unlikely) or the assumption at
> the start was incorrect that the first 3 steps are behaving [...]

It looks like a bearing I've printed sewveral times, so its now warming 
up to print via usb.  That warmup is slow, the ender 5 plus's big plate 
takes a while to get to 80C, so it will be about 1.5 hours.

And unlike many previous attempts, both mc and cura seem to be behaving 
correctly, reporting the same, apparently valid file sizes, where 
previously the file was shown as 0 length but loaded into cura, and 
printed just fine.

> What in the world could I do with this? Cura is not responsible for
> making sure that sshfs is working correctly on your computer. Am I
> making a wrong assumption somewhere? Am I summarising your
> reproduction steps incorrectly? What did I do wrong?

Nothing that I can see from here, and thanks muchly for going to the 
trouble chasing a ghost.

Since it works anyway, can we blame it on Murphy? Or a filesystem that 
doesn't refresh its cache when it needs to? In the latter case, then 
sshfs seems to be the culprit.  But this morning we lack failure 
evidence as I'm failing to verify the problem also. Both machines have 
been updated since I posted the original fuss, perhaps something has 
been fixed?  IDK.

Thank you, Ghostkeeper, I'll advise when the print is done as to the 
success of the print. I did not refresh the beds coat of layerneer so 
slippage could occur.

Cheers, Gene Heskett
-- 
"There are four boxes to be used in defense of liberty:
 soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author)
If we desire respect for the law, we must first make the law respectable.
 - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/gene>

WOWElectronics, 2021-05-21:
On Friday 21 May 2021 05:47:15 Ghostkeeper wrote:

> I will start here with step 4 since the first 3 steps are not in Cura.
> So I must assume that the rest of your computer is in working order.
> So basically the reproduction steps become then (to [...]
Will not complete the load into cura 4.9.1 on this stretch machine,

Shows this as cura appimage is launching and loading untitled.stl:
***@***.***:~/AppImages$ cura
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated 
in favour of importlib; see the module's documentation for alternative 
uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is 
created.
Fontconfig warning: "/etc/fonts/fonts.conf", line 100: unknown 
element "blank"
Cyclic dependency detected 
between "file:///tmp/.mount_curaCMbLEv/usr/bin/resources/qml/Actions.qml" 
and "file:///tmp/.mount_curaCMbLEv/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - 
fallback: "/tmp/.mount_curaCMbLEv/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation
Only C and default locale supported with the posix collation 
implementation
Only C and default locale supported with the posix collation 
implementation
Case insensitive sorting unsupported in the posix collation 
implementation
Numeric mode unsupported in the posix collation implementation

The second time I loaded in tnto cura it did succeed, and did slice it, 
saving the file to that machine. Except it didn't save to that machine, 
it saved it to the same place on this machine as a just sy of 10 
megabyte file.
But this machine is stretch, the target machine is buster.
So I copy that 300 some megabyte untitled.stl to the target machine.
And it shows as a 368some megabyte file there, from here.

So now I goto that machine, and launche cura, again from the AppImage,
and getting the same warnings as above, but it loads, runs and is now 
loading that copy. On a 16gigabyte, 4 core i5 machine that takes several 
minutes.

So I slice it, get 6620894 bytes of gcode.  Saved to disk, read by mc.

> 4.9.1. 3. Let it slice and print via USB cable.
> 4. Save the g-code to disk.
> 5. Verify the file size of the .gcode file created on disk.
>
> I can only send it to an emulator in step 3, since I don't [...]

Moderatly so, I may be set for higher resolution.

> So that means that one of my other assumptions must be wrong. Either
> the printer is not behaving correctly (unlikely) or the assumption at
> the start was incorrect that the first 3 steps are behaving [...]

It looks like a bearing I've printed sewveral times, so its now warming 
up to print via usb.  That warmup is slow, the ender 5 plus's big plate 
takes a while to get to 80C, so it will be about 1.5 hours.

And unlike many previous attempts, both mc and cura seem to be behaving 
correctly, reporting the same, apparently valid file sizes, where 
previously the file was shown as 0 length but loaded into cura, and 
printed just fine.

> What in the world could I do with this? Cura is not responsible for
> making sure that sshfs is working correctly on your computer. Am I
> making a wrong assumption somewhere? Am I summarising your
> reproduction steps incorrectly? What did I do wrong?

Nothing that I can see from here, and thanks muchly for going to the 
trouble chasing a ghost.

Since it works anyway, can we blame it on Murphy? Or a filesystem that 
doesn't refresh its cache when it needs to? In the latter case, then 
sshfs seems to be the culprit.  But this morning we lack failure 
evidence as I'm failing to verify the problem also. Both machines have 
been updated since I posted the original fuss, perhaps something has 
been fixed?  IDK.

Thank you, Ghostkeeper, I'll advise when the print is done as to the 
success of the print. I did not refresh the beds coat of layerneer so 
slippage could occur.

ps:

The print was fine. But out of curiosity, since my .stl's for a similar 
print are about 1% of that 368+ megs, what program did you make 
that .stl with? I'm using openscad here, and the .scad file that src's 
it is rarely over 5k and more comments than actual code. .stl's give or 
take a hundred K.

Cheers, Gene Heskett
-- 
"There are four boxes to be used in defense of liberty:
 soap, ballot, jury, and ammo. Please use in that order."
-Ed Howdershelt (Author)
If we desire respect for the law, we must first make the law respectable.
 - Louis D. Brandeis
Genes Web page <http://geneslinuxbox.net:6309/gene>

closed 2022-01-10 by no-response[bot]