Ultimaker/Cura #9722: cura 4.9.0 can not import materials, they imort but don't show up in the materialslist.

Gignoot, 2021-04-29:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

creality ender 5

### Reproduction steps

* add materials from marketplace
* restart cura
* new materials are nowhere to be found.

### Actual results

no results. materials plugins show as installed , but the materials are nowhere to be found inside cura materials selection.

### Expected results

the materials should be available within the materials list.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

imported materials can not be found inside the cura directory anywhere either. It's a bogus feature as it is.

fvrmr, 2021-04-30:
Hi @Gignoot thank you for your report. 
Could you give me more information about what materials you want to import?

Gignoot, 2021-04-30:
Hi there,

I might have been mistaken, but it could be because I was not logged into my ultimaker account. Though other plugins do install and work without being logged in to account.

The materials I was trying to import were Formfutura. Mainly the rPET.

Kind regards,

Guido.

From: Fenne
Sent: Friday, April 30, 2021 14:05
To: Ultimaker/Cura
Cc: Gignoot; Mention
Subject: Re: [Ultimaker/Cura] cura 4.9.0 can not import materials, they imortbut don't show up in the materialslist. (#9722)

Hi @Gignoot thank you for your report.
Could you give me more information about what materials you want to import?

fvrmr, 2021-05-06:
I can import that material even if I am not logged in. Does it work now for you?

Gignoot, 2021-05-06:
Hello,

Nope , it doesn’t work, it does get downloaded , at this point it gets placed into the following path I found   :    C:\Users\Yadayada\AppData\Roaming\cura\4.9\materials\ReForm_rPET\formfutura_reform+rpet.xml.fdm_material        (it would appear the \Reform_rPET\ folder part is too much.)
I then have to manually import from cura and find this directory to import it from.   
After importing through material manager, a copy of the file is found one directory higher , thus at   C:\Users\Yadayada\AppData\Roaming\cura\4.9\materials\formfutura_reform+rpet+%232.xml.fdm_material
And at this point I can restart cura or not, the material will NOT show up inside cura material manager or anywhere else at all.

There is also the following cura directories where no downloaded materials get found : C:\Users\Yadayada\AppData\Local\cura\4.9\   and     C:\Users\Yadayada\AppData\Local\Ultimaker B.V\Cura\   nor     C:\Program Files\Ultimaker Cura 4.9.0\
I hope this helps things clear up, to me it appears the plugin just has one to many directory in it’s install path ?? Or just the wrong install path all together ??  I installed Cura 4.9 on a clean windows install, so no previous versions of cura have run and so no registry keys from those older versions should be present either.

Kind regards

Sent by Old Fox.

From: Fenne
Sent: Thursday, May 6, 2021 16:25
To: Ultimaker/Cura
Cc: Gignoot; Mention
Subject: Re: [Ultimaker/Cura] cura 4.9.0 can not import materials, they imortbut don't show up in the materialslist. (#9722)

I can import that material even if I am not logged in. Does it work now for you?

Gignoot, 2021-05-07:
Hi there,

That is very odd, since the same material gets sold on both diameters of spool material, I have an rPET 1.75 spool from Formfutura. I have several materials of them, like PP and TPU. All are sold in both diameters.

The diameter does not change anything else about the material property either, both diameters have the exact same material property, so there should not be any problem at all , no matter what diameter gets used. 
It also doesn’t matter for other materials. No it is not a compatibility problem, at least not in real life, my printer prints this material just fine, lol.
I looked at the material file and it says 2.85 only yes, maybe I can just add another similar line mentioning 1.75 aswell. Ehm nope, doesn’t work either after reimporting nothing to be found.

What did work is just replacing the <diameter>2.85</diameter> with a <diameter>1.75</diameter> and then the material shows up in my materialslist for my printer. Succes !!

Greetings.

Sent by Old Fox.

From: Fenne
Sent: Friday, May 7, 2021 16:33
To: Ultimaker/Cura
Cc: Gignoot; Mention
Subject: Re: [Ultimaker/Cura] cura 4.9.0 can not import materials, they imortbut don't show up in the materialslist. (#9722)

Ah I know what is causing this. The profiles you want to import are not compatible with your printer. Your printer uses 1.75mm and the materials are for printers with 2.85mm.

Ghostkeeper, 2021-05-13:
I would consider this a duplicate of https://github.com/Ultimaker/Cura/issues/6236.