Ultimaker/Cura #9723: Using Gradual Support Infill Steps Generates GCode which extrudes multiple times over the same spot

arjeousski, 2021-04-29:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

CR-10 Mini

### Reproduction steps

I've trying to setup my support interface to be similar to raft as described in https://github.com/Ultimaker/Cura/issues/9192. 
I wanted my support surface to be on top of similar structure as raft, so that some of the material from support surface does down into it therefore making it easier to break away support surface from the model.

So I've setup relatively high support density, and tried to use gradual infill steps to only have that density near the support surface.

1. Set Support density to high number ~80%
2. Set Gradual Support Infill Steps to number 3
3. Slicing creates GCODE which goes over extruding the same place multiple times

### Actual results

Generated GCODE, extrudes over the same spot multiple times
[image]

The edge of the support has alot of material which also affects the bottom of the model
[image]

[image]

### Expected results

Ensure that there is no extrusion over the same spot multiple times such that support does not go into the model.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Support Test Project + STL.zip](https://github.com/Ultimaker/Cura/files/6398965/Support.Test.Project.%2B.STL.zip)

fvrmr, 2021-04-30:
Hi @arjeousski thank you for your bug report. 
I will discuss this with the team. Keep you posted!

jellespijker, 2021-05-01:
@fvrmr I think this a duplicate (couldn't find with a quick search) or it might be on our backlog in Jira already

fvrmr, 2021-05-03:
Couldn't find the duplicate, but it we have a ticket on our backlog indeed. 
Developers see CURA-7848

closed 2024-12-09 by github-actions[bot]