Ultimaker/Cura #9725: Re-Probe Bed Between Objects in "One at a time" mode

GoremanX, 2021-04-29:
### Is your feature request related to a problem?

On my very large delta printer, when printing multiple objects "One at a time", every subsequent object is started at a slightly wrong Z height due to expansion of the printer's frame and other components. Depending on how long the first object takes to complete, this is typically a very small difference (like 0.1 to 0.2mm). It's not enough to affect the dimensions of the printed objects, but it's enough to be devastating to the first layer on the next object so that the entire print job can fail at any point after the first object has been completed. This basically renders "One at a time" mode useless on some printers.

### Describe the solution you'd like

A quick probe at the location where an object is about to be printed would allow the firmware to compare the expected vs actual current Z height and apply an adjusted Z-offset for the duration of that object's print time, thereby ensuring the first layer is accurately positioned. This would require the following options:
- retract filament before Z probe (to avoid oozing at the nozzle for probes that tap the nozzle)
- probing nozzle temperature (to prevent a hot nozzle from damaging the build surface for probes that tap the nozzle)
- turn off bed heater during probe (for some non-contact probes that are affected by the electrical field of the bed heater)

### Describe alternatives you've considered

My current solution is to pre-heat the printer for multiple hours before starting the print job, but this is extremely wasteful and sometimes doesn't even work depending on changing conditions throughout the print job.

### Affected users and/or printers

Large printers, especially very large delta printers with long horizontals and verticals and effector rods. These are precisely the types of printers that are useful with "One at a time" for large batches of objects.

fvrmr, 2021-05-03:
Hi @GoremanX thank you for your feature request. 
I have to defer this request because it is not relevant for Ultimaker printers.
But there is a work around that you could use. 

Use the post processing script search and replace. 
Search for `;LAYER:0` and replace with `;LAYER:0\nG10\nM104 S100\nM140 S20\nG30` as an example.

When printing one at a time starting a new model it starts with ;LAYER:0 in the gcode. Replacing that will add these code to your gcode. See this snippet: 
```
;LAYER_COUNT:98
;LAYER:0
G10
M104 S100
M140 S20
G30
M107
M204 S625
M205 X6 Y6
G0 F8571.4 X71.718 Y57.967 Z0.27
M204 S500
M205 X5 Y5
;TYPE:SKIRT
```

Let me know if that works for you!

GoremanX, 2021-05-04:
Thank you, that's a great idea. I guess I could also write a plugin, so the options show up under the "One at a time" settings in Cura.

Question, in the case of a post-processing script, how would I make it take effect starting only at the second object? Or put another way, how would I ignore the first instance of ;LAYER:0 in a file?

closed 2023-07-13 by github-actions[bot]