Ultimaker/Cura #9728: Lost Low Quality Profile 4.9

jxer, 2021-04-30:
### Application Version

4.9.0

### Platform

windows 10

### Printer

ender3 v2

### Reproduction steps

opened program tried to find low quality profile

### Actual results

low quality is no longer there

### Expected results

find low quality

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Ghostkeeper, 2021-04-30:
The Low Quality profile is still there as far as I can see:
[image]

The set of profiles that are available depends on the printer, the nozzles and the material types that you have active. Going by your list of custom profiles there, could it be that you are printing with TPU? Indeed, TPU only has a profile for Super, Dynamic and Standard. I'm guessing that the people who made that profile found that TPU couldn't extrude material fast enough to print realistically with 0.28mm layer height.

Is that your problem? I'm guessing that there are more reproduction steps than what you wrote down.

jxer, 2021-04-30:
Thank you that was it. i was printing in TPU.

How does your .6  layer profile work? any issues?

closed 2021-04-30 by jxer

Ghostkeeper, 2021-05-13:
> How does your .6 layer profile work? any issues?

It's something that came along with project files I loaded in from other issues here on Github. People put those project files in their issue reports so that we can reproduce their slicing issues. I can't actually print them, so I can't tell you how well it works.