Ultimaker/Cura #9731: Cura Not Syncing

tefracky, 2021-04-30:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Install Cura
2. Add a printer and a custom filament
3. Try to sync with another PC

### Actual results

Cura is not syncing. After restart, there is a triangle in the top right corner. After start syncing manually, it shows "Account synced", but the profiles not appear at the other PC.

### Expected results

Cura should sync.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6403952/cura.log)

fvrmr, 2021-04-30:
Hi @tefracky thank you for your report.
You can't sync custom materials, only if you've downloaded them from the marketplace. 
A work around for this is creating a backup and then restore this backup to the other pc. Let me know if that works for you.

tefracky, 2021-04-30:
Thanks a lot, I didn't now! Do you know if printers or profiles sync?

Ghostkeeper, 2021-04-30:
Only the packages that you have installed via marketplace.ultimaker.com will sync. These packages contain material profiles or plug-ins. Printers and custom profiles don't sync via the account.

You can create back-ups of your complete configuration though (under Extensions -> Back-ups). The back-ups can then be restored on a different computer, effectively transferring your configuration to a different computer.

closed 2021-04-30 by Ghostkeeper