Ultimaker/Cura #9733: 4.9.0 Crashing back to desktop

SaberRider666, 2021-04-30:
### Application Version

4.9.0

### Platform

Windows 10 Pro 64bit; AMD Ryzen 5 3600; 32 GB RAM, GeForce GTX 1060 

### Printer

Ender 3 Pro

### Reproduction steps

1. Opened Cura after Update to 4.9.0
2. Clicked on "File", then "Open File"
3. Contour of "open" Window becomes visible

### Actual results

Crashed back to Desktop without further message

### Expected results

Window should open to select file for opening

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6404933/cura.log)

SaberRider666, 2021-04-30:
[image]

fvrmr, 2021-04-30:
Hi @SaberRider666 thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/9668 that's why I will close this issue. 
This bug has to do with Chitubox you can find there how to fix this.

closed 2021-04-30 by fvrmr