Ultimaker/Cura #9734: Dropdown menus for Sending to Print options not working

dotdash32, 2021-04-30:
### Application Version

4.9.0

### Platform

MacOS 11.2.3

### Printer

Ultimaker 3

### Reproduction steps

1. Slice Model
2. Print over network
3. cannot change which of several printers it will send to
Dropdown options are greyed out, cannot select any options.  The same issue happens on 3D PrinterOS plugin, cannot select which printer type to use.

### Actual results

Printer selection options are greyed out, can open drop down menu but cannot select any options.

Screenshots below are from 3D PrinterOS plugin since my work has the Ultimakers that I don't have access to at the moment.  I had the same behavior from the "Print Over Network" option for the Ultimakers themselves.

### Expected results

Can select any option in box.

I downgraded to 4.8.0 and it worked fine.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Images of drop down (again don't have access to Ultimakers for Print over Network at the moment)

[image]

[image]

fvrmr, 2021-05-03:
Hi @dotdash32 thank you for your report.
Do you have a M1 chip? If so we are aware of this issue.

Also good to know the issue with the plugin is not something we maintain.

dotdash32, 2021-05-03:
I have a 2015 MBP with an Intel i7.

I am aware that you all do not maintain the 3DPOS plugin, but I was getting the same behavior at my work's pod of 6 Ultimaker 3/3 extended hooked up to Cura Connect.  The "Print Over Network" option did not let me change which printer to send to, it only had First Available selected. I will try to get some screenshots on my next shift (tomorrow).

fvrmr, 2021-05-04:
Thanks, I have discussed it with the team and I created a ticket on our backlog.
Developers see CURA-8214

dotdash32, 2021-05-05:
Might be a bit late, but screenshots of Ultimaker relevant stuff.   Pod is 6 UM3's.

Thank you!

[image]

[image]

closed 2024-12-09 by github-actions[bot]