Ultimaker/Cura #9735: wrong de Translation

deBugger-, 2021-04-30:
### Application Version

4.9.0

### Platform

linux

### Printer

Anycubic I3 Mega 

### Reproduction steps

Dear all,

Switch to german i18n and have a look on the main menu bar,

You'll see

"Datei", "Bearbeiten", "Ansicht", "Einstellungen" (1st), "Erweiterungen", "Einstellungen" **(2nd)**, "Hilfe"

There are 2 times Einstellungen, where the second one leads to "Cura konfigurieren ...". This is confusing during in discussions, writing manuals and even help each other.

Please rename the second one to "Konfiguration".  Please note: My native language is German. In this context is the word   "Konfiguration" the best choice.

 
[image]

Thank you
 

### Actual results

Main navigation bar which displays 2 equal navigation entry which are leading to different submenus.

### Expected results

A german main navigation instad 
"Datei", "Bearbeiten", "Ansicht", "Einstellungen", "Erweiterungen",  **_"<< Einstellungen >>"_**, "Hilfe"
this
"Datei", "Bearbeiten", "Ansicht", "Einstellungen", "Erweiterungen",  **_"<< Konfiguration >>"_** , "Hilfe"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No logfile needed.

fvrmr, 2021-05-03:
Hi @deBugger- thank you for your report. It is changed, see: https://github.com/Ultimaker/Cura/commit/7c954fdc2f85eb5caf0131739215b1872e3f69de

closed 2021-05-13 by Ghostkeeper

deBugger-, 2021-06-09:
Dear fvrmr,
Thank you. Please note a update from 4.8.? to 4.9.1 is done. But it seems to be that your work is not included. On my machine (Archlinux) there are still 2 times "Einstellungen" in the main Navigation (de language). Your fix is missing.

Can you confirm this?

fvrmr, 2021-06-15:
Thank you for reaching out and it is not in 4.9.1 but it will be included in the next release.