Ultimaker/Cura #9737: Cura 4.9.0 crashes on Toshiba Chromebook 2015

mettleh3d, 2021-05-01:
### Application Version

4.9.0

### Platform

Linux

### Printer

Ender 3 Pro

### Reproduction steps

Cura 4.9.0 will crash when

1. Load any .stl file
2. Highlight model
3. Click resize/rescale button
4. Cura will crash

Note: Cura 4.8.0 will not crash with the same steps

### Actual results

Cura 4.9 crashes

### Expected results

Can resize model as expected

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[Curalog.txt](https://github.com/Ultimaker/Cura/files/6408944/Curalog.txt)

fvrmr, 2021-05-03:
Hi @mettleh3d thank you for your report. 
It looks like this is a duplicate of https://github.com/Ultimaker/Cura/issues/9709
I will close this issue since it is a duplicate.

closed 2021-05-03 by fvrmr

mettleh3d, 2021-05-03:
@fvrmr Thanks!