Ultimaker/Cura #9738: FLsun QQ-s Model Not Correct on screen

JohanPluister, 2021-05-01:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

FLsun QQ-s

### Reproduction steps

If you choose the FLsun qq-s for printer, the printer is turned 90 degrees. This is shedding on the screen.

### Actual results

[image]

### Expected results

Can it be changed again at 4.9.1 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I would like to see that change

GregValiant, 2021-05-01:
This is from your printer definition file in "...\Ultimaker Cura 4.x.x\resources\definitions"
"author": "Cataldo URSO & Eddy Emck ",
        "manufacturer": "FLSUN",
        "platform": "flsun_qq_s.3mf"
The platform "flsun_qq_s.3mf" is in "...\Ultimaker Cura 4.x.x\resources\meshes".  I suppose you could alter that file by rotating the platform in MS Builder or something similar.  In the definition file, if you leave the platform as an empty string **"platform: ""** then you would see just the build plate in Cura.

fvrmr, 2021-05-03:
Hi @JohanPluister thank you for your report. This is a duplicate of #9687 
You could try the work around @GregValiant is suggesting or mine here: https://github.com/Ultimaker/Cura/issues/9687#issuecomment-826850461
I will close this issue since it is a duplicate. But please reach out if you have more questions.

closed 2021-05-03 by fvrmr